OpenDroneMap/ODM #879: theory of producing the orthophoto

xialang2012, 2018-08-27:
Hello, what is the theory used in the ortho module? I seems orthophoto was produced from an textured 3D mesh. @dakotabenjamin @smathermather

pierotofy, 2018-08-28:
Hey @xialang2012 :hand: does this help? http://community.opendronemap.org/t/where-can-i-find-background-information-on-the-concepts-of-odm/665/2

xialang2012, 2018-08-28:
Thank  you @pierotofy, I know the processing steps you presented.
 What I am not sure is whether we need to build DEM first when create the orthophoto. I found some materials which indicated build the DEM is the key factor to generate precision orthophoto. However, after reviewed the source codes of ortho module, I did not find any information about the DEM. It seems that DEM is not necessary in the module. I did not have enough knowledge about the 3D image, so I gave an post here.

Komzpa, 2018-09-11:
@xialang2012 DEM is needed only when you have single shot. On multiple overlapping shots it's possible to deduce 3D position of an object without supplementary DEM. If you're reading oldschool books on it, it would be near "getting DEM from stereo imagery".

smathermather, 2018-09-12:
@xialang2012, @Komzpa is correct. We are deriving a digital surface model automatically from image overlap in order to generate the orthophoto.

smathermather, 2018-09-12:
@xialang2012 -- I would also look to this: http://community.opendronemap.org/t/resources-to-learn-photogrammetry-and-computer-vision/63 for additional context. I am going to close the issue, but if you'd like to discuss more, please feel free to reopen (and let us know how you get on with the readings at https://community.opendronemap.org).

closed 2018-09-12 by smathermather