OpenDroneMap/ODM #883: Output georeferencing error statistics

pierotofy, 2018-08-31:
A user mentioned it would be useful to have odm_georef output the georeferencing error value, so that georeferencing accuracy can be measured. I think we already calculate, we just don't output it.

pierotofy, 2018-09-03:
- When using GCPs this value is in `odm_georef`
 - When not using GCPs, this value can be computed from OpenSfM's reconstruction.json file, which contains a "reprojection_error" field for each observation. Returning the mean of all observations would probably be sufficient for report purposes.

gastgit, 2020-06-14:
> * When using GCPs this value is in `odm_georef` (`minTotError` variable).
> * When not using GCPs, this value can be computed from OpenSfM's reconstruction.json file, which contains a "reprojection_error" field for each observation. Returning the mean of all observations would probably be sufficient for report purposes.

I have looked at the reconstruction.json file after an ODM run at opensfm/reports and at /opensfm and these files do not have the field "reprojection_error". I used ODM 1.0.0.
Was this field removed?

pierotofy, 2020-06-14:
`odm_georef` is no longer in charge of alignment, so this issue should actually be closed.

Any statistics about georeferencing might be exposed via OpenSfM (if any). Not really sure if they are.

closed 2020-06-14 by pierotofy