OpenDroneMap/ODM #886: Read initial photo angles from EXIF

Komzpa, 2018-09-03:
DJI Mavic Pro (at least) stores all the angles of photo in EXIF. 

Gwenview is able to show them:

[image]

It would be great if this info can be reused.

pierotofy, 2018-09-03:
Yes, the problem is that the GPS and compass information are far too inaccurate to skip the structure from motion process, so even if the gimbal information is fairly accurate, we would still need to compute everything else to compute a good reconstruction.

That's to my knowledge at least.

Komzpa, 2018-09-05:
@pierotofy I see the following possibilities for using this data:
 - Get a rough view box of image projected on the ground. Look for pairs by matching overlapping boxes;
 - Remove photos with sky from reconstruction before even starting;
 - Mask sky on photos before finding matching points;
 - Remove photo from reconstruction if computed angles don't match the ones in EXIF.

Does this make sense?

Komzpa, 2018-09-05:
Here's a way to quickly group photos into down-facing and not down facing:
```
exiftool -n -"Directory<GimbalPitchDegree" .
```

After that you get a bunch of directories named as pitch degree. Pull only the ones with <-60 into processing to get rid of the ones having horizon in them, or <-80 to get down-looking only.

pierotofy, 2021-12-23:
This is now available in `2.7.0`! Choose `--sfm-algorithm triangulation` to use OPK angles.

closed 2021-12-23 by pierotofy