OpenDroneMap/ODM #887: Ortophoto improvements

Komzpa, 2018-09-04:
OpenDroneMap Blog features a post on improvements, showing cool perpendicular buildings:
https://www.opendronemap.org/2018/07/repost-upcoming-improvements-to-opendronemap-better-everything/

However Docker latest version still produces this:
[image]

Where do I take the version featured in blog post?

pierotofy, 2018-09-05:
The version featured on the blog hasn't been released as of yet. We don't have a timeline, but we're hoping to release it within a few weeks. :+1:

Komzpa, 2018-09-05:
@pierotofy is there a branch to look at?

pierotofy, 2018-09-05:
Some (not all) work is available on: https://github.com/OpenDroneMap/OpenDroneMap/commit/8e59b2f107f02b6b119f73885801d691dea99f19

I don't think it's in usable state however.

hoffa111, 2018-09-12:
@pierotofy Hello,
Is it possible to use v0.4 RC1 to get better results for orthophoto? I built it from source on my ubuntu and it still produced results with warping and visual artifacts. Thanks in advance.

pierotofy, 2018-09-12:
@hoffa111 make a clone from `master`. We just merged #889 :+1:

hoffa111, 2018-09-12:
@pierotofy many thanks, I will try! Which parameters I need to use to get better quality? In previous version I used those parameters:
--resize-to: -1
--min-num-features: 25000
--mesh-octree-depth: 6
--mesh-solver-divide: 6
--orthophoto-resolution: 100
--dem-terrain-type: FlatNonForest
--force-ccd: 13.1
Also, I mainly use it for photos with buildings.

pierotofy, 2018-09-12:
Change `--orthophoto-resolution` to 0.001 with the new version (we've changed the units), also remove --mesh-octree-depth and --mesh-solver-divide.

hoffa111, 2018-09-13:
@pierotofy Hello,
I just tested it and roofs of buildings has strange wavy effects (see attachment). I used these parameters:
--min-num-features: 25000
--orthophoto-resolution: 0.001
--force-ccd: 13.1
How can I fix this? Thanks in advance.

[image]

pierotofy, 2018-09-13:
Let's move the conversation over to https://community.opendronemap.org since this is starting to look like a question thread and the GitHub repo should be used only for bug reports. Thank you!

closed 2018-09-13 by pierotofy

Komzpa, 2018-09-16:
The link won't work:
[image]

pierotofy, 2018-09-16:
Ah, sorry, use this URL (non-https): http://community.opendronemap.org