OpenDroneMap/ODM #888: Syntax error in configure.sh on Ubuntu 16.04 / ODM Version 0.4 RC1

ichfred, 2018-09-09:
### How did you install OpenDroneMap? (Docker, natively, ...)?

natively, on a fresh installed Ubuntu 16.04. Using the latest version 0.4 RC 1.zip file.

### What is the problem?

Syntax error while running configure.sh:

xxx@xxx01:~/ODM$ bash configure.sh install
configure.sh: line 2: $'\r': command not found
configure.sh: line 9: syntax error near unexpected token `$'{\r''
'onfigure.sh: line 9: `install() {

pierotofy, 2018-09-09:
Does it work if you make a git clone from master instead of using the zip file?

ichfred, 2018-09-10:
No problems with a git-clone.

pierotofy, 2018-09-10:
@dakotabenjamin could this be caused by Windows new lines?

dakotabenjamin, 2018-09-10:
Yes, the problem was with the compressed files having come from windows. I just fixed them!

closed 2018-09-10 by dakotabenjamin