OpenDroneMap/ODM #893: ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

grumpydev-net, 2018-09-15:
### How did you install OpenDroneMap? (Docker, natively, ...)?
Docker for Linux

### What's your browser and operating system? (Copy/paste the output of 
https://whatismybrowser.com/w/5GWYRKK

### What is the problem?

Error execution the following command:
 sudo docker run -ti --rm -v ~/Documents/Mapping:/datasets/code opendronemap/opendronemap --project-path /datasets
Traceback (most recent call last):
  File "/code/run.py", line 8, in <module>
    import ecto
  File "/code/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory'

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
Just start the processing of the photos

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

just fresh docker pull opendronemap/opendronemap
Type a command equivalent to:
docker run -ti --rm -v ~/Documents/Mapping:/datasets/code opendronemap/opendronemap --project-path /datasets

pierotofy, 2018-09-15:
Mm, I wonder if something didn't build properly @smathermather @dakotabenjamin ?

I've just ran the latest build `610be539ceb9` and that seemed to work fine. Run `docker pull opendronemap/opendronemap` and see if that fixes the problem.

pierotofy, 2019-05-23:
Good news, ecto is now gone! So this problem should also be gone with `0.6.0`. Please update and see if things work OK. If not, please open a new issue. :+1:

closed 2019-05-23 by pierotofy