OpenDroneMap/ODM #897: Support for external georeferenced image list

kikislater, 2018-09-21:
### What is the problem?

Related to this : http://community.opendronemap.org/t/how-to-prepare-photos-with-location-information-for-webodm/1106
 
ODM currently support georeferenced images from geotags present in exif. RTK/PPK process generate an external text/csv file with filename and llh (known in survey with latitude, longitude, height) or ECEF (cartesian coordinates) or local coordinates. Some users with other thing than Dji doesn't know how to process txt file to geotag image (usage of exiftool for example). Selecting a text file and images without embeded geotag to process dataset should be a good  option.
Furthermore as explained in this issue : https://github.com/OpenDroneMap/WebODM/issues/282 , standard geotag could not achieve ppk/rtk precision and accuracy. External file with embeded could solve this issue.

I could provide dataset with PPK file.

As explain in process here : http://community.opendronemap.org/t/where-can-i-find-background-information-on-the-concepts-of-odm/665/2 it could skip one step. 

Others photogrammetry do it like : 

- MicMac : Used in "OriConvert OriTxtInFile" process => https://micmac.ensg.eu/index.php/OriConvert
File structure is describe as 
#F=N Y X Z K W P
image latitude longitude altitude yaw pitch roll

- Photoscan : Import txt or csv in reference pane georeference photos
[image]

- Pix4d :  
https://support.pix4d.com/hc/en-us/articles/202558539-Input-files#labelA1_1

Note : Could be also in part of project we presented in FOSS4G 2018 Dar Es Alam with @juldebar @sbonhommeau : https://github.com/juldebar/Deep_mapping

pierotofy, 2019-04-09:
For future reference, OpenSfM has a way to override the EXIF GPS information via a `exif_override.json` file (which could be useful for this).

```
{
    "image_name.jpg": {
        "gps": {
            "latitude": 52.51891,
            "longitude": 13.40029,
            "altitude": 27.0,
            "dop": 5.0
        }
    }
}
```

gastgit, 2020-03-23:
> For future reference, OpenSfM has a way to override the EXIF GPS information via a `exif_override.json` file (which could be useful for this).
>
> ```
> {
>     "image_name.jpg": {
>         "gps": [...]

Tnx for your reply @pierotofy . Can you point to instruction on where such json is used in webODM in order to improve accuracy of geotag?
Appreciate the help.

pierotofy, 2020-03-23:
WebODM currently does not have support overriding georeferencing info. We'd welcome contributions toward that. Interested to help @gastgit ?

gastgit, 2020-03-24:
Tnx @pierotofy so ODM command line has this option?
Please contact me offline for contributing.

pierotofy, 2020-03-24:
The command line doesn't explicitly have this option, but it can be "hacked" by adding an exif_override.json file in the OpenSfM directory.

You can use https://uav4geo.com/contact for private conversations.

gastgit, 2020-03-24:
Hi,
While running using webODM, the EXIF GPS data was as below with apparent coordinate accuracy is on the order of some meters
[image]

But , using this page [https://www.opendronemap.org/2020/02/checking-a-running-process-in-webodm/](url) I was this which displays  coordinate with accuracy on the order of nano-meters... what am I missing? What is the actual accuracy I will get from webODM?
Tnx
[image]

pierotofy, 2020-03-24:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). :+1:

pierotofy, 2020-09-14:
This is implemented in #1156 :clinking_glasses:

smathermather, 2020-09-15:
Ha! I missed this.

closed 2020-09-15 by smathermather

kikislater, 2020-09-17:
Cool ! Thank you @pierotofy