OpenDroneMap/ODM #899: --use-25dmesh not working

TheDude005, 2018-09-27:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 69 on Linux  Ubuntu 18.04.1 LTS

### What is the problem?

2.5d mesh flag is failing in run.py

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

That run.py would accept the --use-25dmesh parameter

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

After reading that a new release came out I was excited to try it out with a dataset that I know works.  I have images in my home directory in a folder called images.  I ran 
`docker run -ti --rm -v /home/Jordan:/datasets/code opendronemap/opendronemap --project-path /datasets`
and received an error about not having enough valid images in /datasets/code/images.  Thinking that I must not have pulled the latest version I ran docker pull opendronemap/opendronemap.  After that was done I ran the command again.  I got the same error about not enough valid images in /datasets/code/images.  So I went into Docker and ran rmi on all of the opendronemap images to make sure I would get a fresh install.  I ran that code again and it downloaded and extracted the new images and still failed with the same error.  At this time I decided to use my runtime parameters from before to see if it would work.
`sudo docker run -ti --rm -v "$(pwd)/images:/code/images" -v "$(pwd)/odm_orthophoto:/code/odm_orthophoto" -v "$(pwd)/odm_georeferencing:/code/odm_georeferencing"  opendronemap/opendronemap`

This time opendronemap ran successfully and outputed the orthophoto.  After looking at the quality I decided to run it again using the 2.5d mesh to see how that affected the quality.  I deleted the two output folders and ran
`sudo docker run -ti --rm -v "$(pwd)/images:/code/images" -v "$(pwd)/odm_orthophoto:/code/odm_orthophoto" -v "$(pwd)/odm_georeferencing:/code/odm_georeferencing"  opendronemap/opendronemap --use-25dmesh`

The reply from the program was

> usage: run.py [options] <project name>
run.py: error: unrecognized arguments: --use-25dmesh

I tried it several times but got the same reply so I was curious if 2.5d mesh was not working right now or if there is a bug where it is not accepting that parameter.

pierotofy, 2018-09-28:
The `--use-25dmesh` parameter does not exist anymore. 2.5D meshes are now the default for all orthophotos. You can use the 3D mesh for orthophoto generation by using `--use-3dmesh` instead. See `./run.sh --help` for up to date parameters.

closed 2018-09-28 by pierotofy

TheDude005, 2018-09-28:
So if I wished to update the documentation how would I go about doing that for y'all or do you want to leave that parameter in the documentation for earlier versions?

pierotofy, 2018-09-28:
We would love to have some help updating our documentation!

You can change it directly by editing https://github.com/OpenDroneMap/docs/blob/publish/source/using.rst (or any other page within the docs repo).