OpenDroneMap/ODM #901: opensfm-depthmap-resolution and orthophoto-resolution options not recognized

plouj, 2018-10-05:
### How did you install OpenDroneMap?

docker build from the current master (002a6f0)

### What's your browser and operating system?
Linux

### What is the problem?

The new options are not recognised:

```
docker run -it --rm \
    -v $(pwd)/images:/code/images \
    -v $(pwd)/odm_meshing:/code/odm_meshing \
    -v $(pwd)/odm_orthophoto:/code/odm_orthophoto \
    -v $(pwd)/odm_georeferencing:/code/odm_georeferencing \
    -v $(pwd)/odm_texturing:/code/odm_texturing \
    -v $(pwd)/opensfm:/code/opensfm \
    -v $(pwd)/pmvs:/code/pmvs \
    odm \
    --end-with mvs_texturing --mesh-size 400000 --mesh-octree-depth 12 --force-ccd 13.2 --texturing-keep-unseen-faces --opensfm-depthmap-resolution 1280 –-orthophoto-resolution 0.0001

run.py: error: unrecognized arguments: --opensfm-depthmap-resolution 1280 –-orthophoto-resolution 0.0001

```

### What should be the expected behavior?

It should work as described on https://www.opendronemap.org/2018/09/better-everything-announcing-opendronemap-0-4/

plouj, 2018-10-06:
I'm not sure what I was doing wrong before, but when I changed it to  ```--orthophoto-resolution 0.0001 --depthmap-resolution 1280``` both options were accepted. Previously even just ```–-orthophoto-resolution 0.0001``` wasn't working.

closed 2018-10-06 by plouj

mcnis003, 2018-10-18:
Sorry to bring this back from the dead, but I have the same behavior.

This fails
`docker run -i
-v /home/ianmcnish/rgb_gcp_list.txt:/code/images/rgb_gcp_list.txt
-v /home/ianmcnish/:/code/images
-v /home/ianmcnish/:/code/odm_orthophoto
-v /home/ianmcnish/:/code/odm_texturing opendronemap/opendronemap
--orthophoto-resolution 3.0
--gcp /code/images/rgb_gcp_list.txt`

Error:
`run.py: error: unrecognized arguments: -–orthophoto-resolution 3.0`

This works
`docker run -i
-v /home/ianmcnish/rgb_gcp_list.txt:/code/images/rgb_gcp_list.txt
-v /home/ianmcnish/:/code/images
-v /home/ianmcnish/:/code/odm_orthophoto
-v /home/ianmcnish/:/code/odm_texturing opendronemap/opendronemap
--orthophoto-resolution 0.0001
--depthmap-resolution 1280
--gcp /code/images/rgb_gcp_list.txt`