OpenDroneMap/ODM #902: Not possible to save the smvs results

plouj, 2018-10-10:
This line:
https://github.com/OpenDroneMap/OpenDroneMap/blob/9b4345681518fc668a2a051e032e3032ce9e747e/scripts/smvs.py#L63 makes it almost impossible to save the smvs step's work in order to be able to resume without having to redo it. Currently running with docker like:

```
docker run -it --rm \
    -v "$(pwd)/images:/code/images" \
    -v "$(pwd)/odm_georeferencing:/code/odm_georeferencing" \
    -v "$(pwd)/odm_meshing:/code/odm_meshing" \
    -v "$(pwd)/odm_orthophoto:/code/odm_orthophoto" \
    -v "$(pwd)/odm_texturing:/code/odm_texturing" \
    -v "$(pwd)/opensfm:/code/opensfm" \
    -v "$(pwd)/smvs:/code/smvs" \
    opendronemap/opendronemap
```

results in:

```
...
[INFO]    Running SMVS Cell
Traceback (most recent call last):  
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)  
  File "/code/scripts/smvs.py", line 63, in process
    shutil.rmtree(tree.smvs)  
  File "/usr/lib/python2.7/shutil.py", line 256, in rmtree
    onerror(os.rmdir, path, sys.exc_info()) 
  File "/usr/lib/python2.7/shutil.py", line 254, in rmtree 
    os.rmdir(path)  
OSError: [Errno 16] Device or resource busy: '/code/smvs'
```

plouj, 2018-10-11:
Yes, the container is mounting the ```/code/smvs``` directory because I told it to. If I don't then I won't be able to save a copy of the intermediate results and continue where I left off (which is pretty important if I want to for example delete some bad vertices before proceeding into meshing or if I'm simply forced to restart because of [issues with running pdal](http://community.opendronemap.org/t/what-determines-the-amount-of-memory-used-by-pdal/1226)). The command I referenced is directly from the ```README.md``` file so I thought it should work.

pierotofy, 2018-10-11:
Yes, I think we should change the command reference in the `README.md`.

Perhaps you could help us and change it if you have time? https://github.com/OpenDroneMap/OpenDroneMap/edit/master/README.md

plouj, 2018-10-14:
I would love to, but I think it would be faster for someone else to do it instead of me at this time.

closed 2020-04-14 by smathermather