OpenDroneMap/ODM #903: failed with error code 1 in /tmp/pip-install-5ThHG1/PyYAML/

droneros, 2018-10-11:
Hi, Im getting this error when trying to start webodm on an ubuntu cloud server... any idea? Thanks!

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-5ThHG1/PyYAML/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-H70vfI/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-5ThHG1/PyYAML/
[root@vps-1541293-x ~/WebODM/WebODM] # ./webodm.sh start

pierotofy, 2018-10-12:
This issue repository is for reporting bugs, this looks like an installation question. Can we move the conversation over to http://community.opendronemap.org? Thank you!

closed 2018-10-12 by pierotofy

droneros, 2018-10-12:
Thanks Piero,

I just created this one:
http://community.opendronemap.org/t/need-help-troubleshooting-why-my-webodm-instalated-on-a-cloud-server-is-not-working/1237
I can provide access to my server if needed.

Thanks,

Adrian.