OpenDroneMap/ODM #907: Georeferencing failing even with geotagged photos

thsant, 2018-11-01:
Using ODM 0.4, I got the following error:

```
Error in OdmOrthoPhoto:
Problem reading header in modelfile!
```

Investigating the issue, I saw that `odm_textured_model_geo.obj` was not generated. Previously, I got the following warning:
`
[WARNING] Georeferencing failed. Make sure your photos have geotags in the EXIF or you have provided a GCP file.`

The photos were produced by a DJI Phantom 4 Pro and, apparently, all of them are properly geottaged:

```
GPS Version ID                  : 2.3.0.0
GPS Latitude Ref                : South
GPS Longitude Ref               : West
GPS Altitude Ref                : Above Sea Level
GPS Map Datum                   : WGS-84
GPS Altitude                    : 917.3 m Above Sea Level
GPS Latitude                    : 21 deg 58' 36.09" S
GPS Longitude                   : 47 deg 51' 10.74" W
GPS Position                    : 21 deg 58' 36.09" S, 47 deg 51' 10.74" W
```
I have gotten this issue over and over again, with different sets of images, since 0.4 update.

Any tip of advice?

pierotofy, 2018-11-02:
Hey @thsant , thanks for reporting the problem.

Could you upload a few of the images for us to take a look at?

thsant, 2018-11-05:
Thanks, @pierotofy. [At Dropbox](https://www.dropbox.com/sh/q3zevsytxp6dtzl/AACQ6VgsCmI6mcIazW6ybcpGa?dl=0) you will a few images plus some files I guess could help :

```
img_list.txt
odm_25dmesh.ply
odm_mesh.ply
reconstruction.json
```
Such files show that structure from motion, multiple view stereo and meshing steps look fine.

The command employed:

`docker run -ti --rm -v /home/odm/tutorial:/datasets/code opendronemap/opendronemap --project-path /datasets --start-with odm_orthophoto --dsm --dtm --orthophoto-resolution 2.0 --build-overviews --proj '+proj=longlat +ellps=WGS84 +datum=WGS84'`

pierotofy, 2018-11-05:
Ah, I think the problem is the `--proj` param. You typically don't want to pass that parameter if you want to use the EXIF coordinates. Could you try to reprocess without it?

thsant, 2018-11-06:
Thanks, @pierotofy, it's working now (removing the `--proj` option).

I (wrongly) assumed `--proj' could be employed to transform data between spatial reference systems.

smathermather, 2018-11-06:
Excellent!

closed 2018-11-06 by smathermather