OpenDroneMap/ODM #910: Problem with a dataset

elgarbe, 2018-11-09:
Hi, I have two datasets from the same place but taking with different techniques. 
I'm running ODM with docker:
docker run -ti --rm -v /my/project:/datasets/code opendronemap/opendronemap --project-path /datasets under ubuntu 16

I obtain good orthoimage with the first dataset, but not with the second one.
I attached a link to them
https://drive.google.com/open?id=1yNys3QkhnaBie62zTviyKUTsdxpdiSu-

can someone tell me what is wrong with the second dataset?

pierotofy, 2018-11-09:
Hey @elgarbe :hand: looks like the model is flipped upside down. 

Looks like this is caused by some of your images which are missing the altitude values.

```
exiftool image_1104180532.jpg
....
Aperture                        : 2.9
GPS Date/Time                   : 2018:11:04 18:05:32Z
GPS Latitude                    : 33 deg 30' 29.32" S
GPS Longitude                   : 60 deg 3' 43.58" W
GPS Position                    : 33 deg 30' 29.32" S, 60 deg 3' 43.58" W
```

```
exiftool image_1104180242.jpg 
...
Aperture                        : 2.9
GPS Altitude                    : 87.9 m Above Sea Level
GPS Date/Time                   : 2018:11:04 18:02:42Z
GPS Latitude                    : 33 deg 30' 28.92" S
GPS Longitude                   : 60 deg 3' 47.10" W
GPS Position                    : 33 deg 30' 28.92" S, 60 deg 3' 47.10" W
```

Notice the GPS Altitude field.

pierotofy, 2018-11-09:
I wonder if we should have better checks sanity checks and catch these problems early in the pipeline.

elgarbe, 2018-11-09:
ok, but in the first dataset I have missing altitude data on some images but the result is good. I don't get a odm_georeferencing becose of this.

how can I put altitude data by hand?

thank!

AnthonyCoughlin, 2019-01-22:
Can ODM process images with no EXIF data at all?  This would be useful if extracting frames\images from video to use for 3D modeling.

smathermather, 2019-01-22:
ODM can process with no EXIF data. It will create a 3D model, but no real orthophoto an no georeferencing.

pierotofy, 2020-09-14:
I think this can be closed:

 - We have warnings for missing GPS tags
 - we have support for pseudo-georeferencing for non-georeferenced datasets.

:+1:

closed 2020-09-14 by pierotofy