OpenDroneMap/ODM #911: Processing report (feature request)

brookjason, 2018-11-13:
Apologys if in the wrong place this is more of a feature request.

Some sot of report similar to this [Trial_2.pdf](https://github.com/OpenDroneMap/ODM/files/2576856/Trial_2.pdf) This is from agisoft photoscan.

smathermather, 2018-11-13:
This is an excellent idea, and there is a skeleton of work done in OpenSfM to help build this out. Thanks for the request! I think there might be an issue out there already either in WebODM or ODM. If you can find that and link these, that would be really helpful.

brookjason, 2018-11-14:
> This is an excellent idea, and there is a skeleton of work done in OpenSfM to help build this out. Thanks for the request! I think there might be an issue out there already either in WebODM or ODM. If you can find that and link these, that would be really helpful.

@smathermather , I was unsure as to which one to put it under (ODM or WebODM) I did have a look and search through the ODM and haven't spotted anything like this. I will have a better look in both and link them if I find anything.

saffa19, 2019-06-21:
This is something I'd really like to see. @brookjason what did you use to generate the graphics for overlap and camera positions/errors?

brookjason, 2019-06-21:
> This is something I'd really like to see. @brookjason what did you use to generate the graphics for overlap and camera positions/errors?

This was automatic output from Agisoft Metashape formally known as Photoscan

pierotofy, 2021-04-05:
This is done! :clinking_glasses:

closed 2021-04-05 by pierotofy