OpenDroneMap/ODM #914: [Feature Request] DTM/DSM Resolution

smathermather, 2018-11-21:
It would be nice to have DSM/DTM resolution based on some multiple of the resolution calculated for the orthophoto, e.g. if ODM calculates that the optimum resolution for orthophoto is 5 cm, default to 20 cm for the DSM and DTM resolutions. This is commonly how closed source packages handle this, and of course, like orthophoto resolution would require a flag for override.

pierotofy, 2018-11-21:
This would be a nice feature. Currently `--dem-resolution` is the same as `--orthophoto-resolution` by default (and both are capped by the GSD estimate ceiling, unless `--ignore-gsd` is used). So you're suggesting that `--dem-resolution` should be a multiple of `--orthophoto-resolution`? Or perhaps it's the interpolation radius that should be incremented? 

See https://github.com/OpenDroneMap/ODM/blob/master/scripts/odm_dem.py#L96 since we currently don't allow a user to specify the radius.

smathermather, 2018-11-22:
More the GSD ceiling, and it looks like the interpolation radius gets modified automatically as the GSD changes, though the radius might be useful to expose, I'm not jumping to add more flags for people... . 

I put in a [pull request](https://github.com/OpenDroneMap/ODM/pull/915) for doing 4x this. I'll do some further testing -- I've got someteen thousand images to test this on in the next couple of months, but this patch should help others while we discover what these values should be. I think point clouds need some improvements before we can really understand how to really improve these, but at least we can prevent people from having memory problems.

smathermather, 2018-12-08:
Closed by https://github.com/OpenDroneMap/ODM/pull/915

closed 2018-12-08 by smathermather