OpenDroneMap/ODM #916: Process killed after a long time in smvs.py

hcwinsemius, 2018-11-22:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker install on Ubuntu 18.04 - 16gig

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 70 on Linux

### What is the problem?

I am processing 538 images. After quite a while the process is killed with the following error message.

Processing view view_0535.mve...
Processing view view_0536.mve...
Processing view view_0537.mve...

Created 538 views with 535 valid cameras.
Imported 535 undistorted images.
[DEBUG]   running /code/SuperBuild/src/elibs/smvs/app/smvsrecon -t2 -a1.0 --max-pixels=409600 -o1 --debug-lvl=0    /code/smvs
Shading-aware Multi-view Stereo (built on Nov  2 2018, 01:48:03)

Initializing scene with 535 views...
Initialized 535 views (max ID is 537), took 36ms.
Reading Photosynther file (538 cameras, 543780 features)...
Automatic input scale: 3
Input embedding: undist-L3
Output embedding: smvs-B3
Running view selection for 535 views...  done, took 290.253s.
Resizing input images for 535 views... Killed
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/smvs.py", line 85, in process
    system.run('%s %s %s' % (context.smvs_path, ' '.join(config), tree.smvs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

I have tried this with --max-concurrency 2 (instead of the default: 8) but this does not help. I don't see any outputs in the designated folder structure. I have 16gb ram and have 233 GB disk space available. Not sure if docker can see this diskspace though.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

The process should end successfully
### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I used the following command: docker run -it --rm -v "$(pwd)/$experiment/images:/code/images" -v "$(pwd)/$experiment/gcp:/code/gcp" -v "$(pwd)/$experiment/odm_dem:/code/odm_dem" -v "$(pwd)/$experiment/odm_orthophoto:/code/odm_orthophoto" -v "$(pwd)/$experiment/odm_georeferencing:/code/odm_georeferencing" opendronemap/opendronemap --gcp /code/gcp/gcp_list.txt --dtm --orthophoto-resolution 10 --dem-resolution 10 --max-concurrency 2

The dataset is quite large so difficult to upload anywhere for testing.

pierotofy, 2018-12-12:
It's a memory problem. You could try to increase your swap space. `smvs` has a pretty steep memory requirement at the beginning of the reconstruction process, but with 16 GB if you allocate 16GB of swap you should be able to make it.

[image]

pierotofy, 2019-06-09:
This should no longer apply as we have switched to using MVE.

closed 2019-06-09 by pierotofy