OpenDroneMap/ODM #919: Exporting Projection Matrix

manik9000, 2018-11-28:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? 

Chrome 70 on Linux

### What is the problem?

Is there any way to export the projection matrix for each used for building pointcloud?

What I am trying to do is that, I want to project a 3D point (from geo-referenced pointcloud) back to image. Is there any way to do it already? If not, can we export camera projection matrix?   

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

[Type answer here]

pierotofy, 2018-11-29:
Hey @manik9000  :hand: see `odm_georeferencing/odm_georeferencing_transform.txt` for the local to geo coordinates transform, `opensfm/reconstruction.json` then has the camera parameters in local coordinates from which you could do the projection.

pierotofy, 2018-12-06:
Linking a relevant discussion: http://community.opendronemap.org/t/rotation-angles-under-reconstruction-json/1431

garlac, 2018-12-07:
You can also get the rotation matrix from opensfm/bundle_r000.out
Refer this link for more info - http://www.cs.cornell.edu/~snavely/bundler/bundler-v0.4-manual.html#S7

manik9000, 2018-12-07:
@pierotofy @garlac thanks for the response guys.

One more question, off the topic..
can you guys point me to some discussion on How can I get denser pointclouds?
Note: I have already gone through an issue talking about exposing few params from OpenSFM.
I tired all the param values suggested there.. but it always produces 1 million points no matter what I try

smathermather, 2018-12-07:
@manik9000 -- a good question for http://community.opendronemap.org, but in short, try this pull request: https://github.com/OpenDroneMap/ODM/pull/921.

I am shamelessly looking for testers of it anyway. :)

manik9000, 2018-12-12:
@pierotofy
I tired your suggestions but they are not producing the correct results.
can you comment on my following workflow:
1. Convert Lat-Lon-Alt to UTM using `proj.txt : +units=m +no_defs=True +datum=WGS84 +proj=utm +zone=15`
2. convert UTM XYZ to geocentric coordinates 
```
def gps_to_ecef(lat, lon, alt):
    ecef = pyproj.Proj(proj='geocent', ellps='WGS84', datum='WGS84')
    lla = pyproj.Proj(proj='latlong', ellps='WGS84', datum='WGS84')
    x, y, z = pyproj.transform(lla, ecef, lon, lat, alt, radians=False)

    return x, y, z
```
3. Use Tgeo (georeferencing_transform), Tcam, and K to project XYZ to pixX, pixY [something like] :
```
a = inv(Tgeo) * (np.matrix([x, y, z, 1])).T
imgPt = inv(Tcam) * a
tt = K * (np.matrix([ imgPt[0]/imgPt[2], imgPt[1]/imgPt[2], 1])).T

Note: I am using cv2.Rodrigues() to get camera Rotation matix
```
Unfortunately, these steps don't work with ODM outputs as expected where as they work excellent with Agisoft's outputs (camera_calibration.xml). 

Am I missing something here? are there any some sort of offsets or something that I have to consider in case of ODM camera calibration?

closed 2019-02-07 by manik9000