OpenDroneMap/ODM #926: Calculating cropping area takes a long time with large datasets

pierotofy, 2018-12-14:
The following command:

```
pdal translate -i odm_georeferenced_model.laz -o "output.filtered.las" decimation outlier range --filters.decimation.step=90 --filters.outlier.method=radius --filters.outlier.radius=2 --filters.outlier.min_k=2 --filters.range.limits='Classification![7:7]'
```

Which is used to calculate a filtered + decimated point cloud (used to compute the convex hull that determines the crop area) is quite slow. Perhaps the sparse point cloud should be georeferenced and used instead, or perhaps a faster tool can be used to estimate a good convex hull.

smathermather, 2018-12-14:
I wonder if a convex hull of the the camera positions would work well. It would solve the problem of orbited objects, but give a generous if somewhat more conservative bounds for the ortho clipping for mapping projects.

pierotofy, 2018-12-14:
That's an interesting approach. Would certainly be faster.

smathermather, 2018-12-14:
Hate to add flags, but a convex hull could be the default behavior, and for folks who are willing to wait, they could use the old approach.

pierotofy, 2018-12-14:
If we don't want flags, we could simply switch approaches based on the input size... large datasets --> cameras, small dataset --> point cloud.

pierotofy, 2018-12-15:
One thing I'm noticing as I'm starting to process larger and larger datasets, is that the defaults we choose tend to work well with smaller datasets, but need some manual tweaking for better results over large datasets. One other example is the `mesh-size` parameter, the default of `100000` works well for smaller areas, but on bigger areas it's flattening a lot of details due to the limited vertex count.

Perhaps at some point in the future we should consider integrating more "smart" guesses and automatically tweak the default parameters based on the input data. This should involve a discussion and methodical evaluation of choices over a number of sample datasets.

smathermather, 2018-12-15:
Have you tested hexbins?: https://pdal.io/stages/filters.hexbin.html

pierotofy, 2019-05-23:
This is much more efficient now that we don't do filtering on the point cloud (it's already done in odm_filterpoints). I think it can be closed for the time being.

closed 2019-05-23 by pierotofy