OpenDroneMap/ODM #928: odm_georef does not handle points clouds > 600 million points

pierotofy, 2018-12-23:
While it doesn't crash, the point cloud it generates is corrupted. This is likely due to some overflow in the PCL PLY parser.

Artifacts:

[image]

How it should look instead:

[image]

I will rewrite the PCL loading code and add the ability to do stream read/writes, so that the point cloud does not need to be loaded in memory all at once.

pierotofy, 2018-12-23:
Or perhaps using the PDAL C++ API.

By using PDAL we could remove the unnecessary step of generating a PLY file and we could go directly to LAZ by passing the SRS as a param to odm_georef.

smathermather, 2018-12-23:
Could speed that step up a bit too.

garlac, 2019-01-10:
I have tested it and found that the geo-referenced point cloud came good. Thank you @pierotofy .

pierotofy, 2019-01-10:
Thanks for testing @garlac!

@dakotabenjamin do you need more time to test #929?

garlac, 2019-01-10:
It was Linux env and native install. 64 Gb RAM 12 cores.

closed 2019-01-12 by pierotofy