OpenDroneMap/ODM #946: OdmExtractUtm Cannot read EXIFs for Mavic 2 Zoom

pierotofy, 2019-02-13:
Trying to process images from the Mavic 2 Zoom causes odm_extract_utm to fail with:

```
Error in OdmExtractUtm:
Failed parsing GPS position for DJI_0076.jpg
```

Related thread with images sample: https://community.opendronemap.org/t/i-can-not-create-an-orthophoto/1759

juvinski, 2019-02-15:
Hi,

I faced the same problem and one of my images was without the GPS Altitude field filled.
Could you please send the exiftool of your image?

pierotofy, 2019-02-15:
If you look at discussion in the forum there's a download link to the images.

juvinski, 2019-02-15:
Sorry, I didn't realize that.
All the images are without the Altitude information - DJI_0073, DJI_0074, DJI_0076, DJI_0077 and DJI_0078.

GPS Altitude                    : 0 m Above Sea Level

juvinski, 2019-02-15:
I was with same problem, but in my case just 1 image of 274 was with value 0 for Altitude - the 273 was right but the orthophoto wasn't generated. In this sample, 6 images of 6 was with without the Altitude information. Maybe a trigger to check the number of images with incorrect GPS data(altitude, latitude and longitude) could be implemented to ensure if you have values less than this threshold this errors could be ignored or not :)

bubi1080, 2019-02-20:
Dear all,

just managed to install WebODM on Mac and am super happy that it is all up and running now. :-) Just ran 109 pics that I took with my Mavic 2 Pro and Drone Harmony App. It took some time to process but finished successfully. The point cloud is there and looks great although seems to be flipped upside down (i.e. the treetops are pointing down in the 3D model. Will post in ODM Community on this issue.

Unfortunately I am having the same issue and cannot create an Orthophoto.

> An orthophoto could not be generated. To generate one, make sure GPS information is embedded in the EXIF tags of your images, or use a Ground Control Points (GCP) file.
> 
All images can be found here (still uploading atm):
[https://1drv.ms/f/s!Ak7jPKJPqpeBgYseINpDzXhvoFbuMQ](url)

The full text from Terminal can be found here: [https://1drv.ms/u/s!Ak7jPKJPqpeBgYspeExTnc3iUlH7OQ](url)

@pierotofy so glad the project exists but is there any idea how the issue can be resolved?

Thanks a million and best regards

bubi1080, 2019-02-21:
Herwith an update, I did manually correct the altitude data in Adobe Bridge from "0m below sea level" to 30m and now it seems to be all working just fine! Herwith link to Task Output: https://1drv.ms/t/s!Ak7jPKJPqpeBgakJ_TT-68SwVLTz4A

Not sure this is a bug in WebODM or rather an issue with the metadata written to the files from the Mavic 2.

juvinski, 2019-02-21:
hi @pierotofy , can I change the default behavior of the utm extract to deal with altitude on the different way of today implementation? Any ideia how to deal with altitude ? what would be the impact of this change on the other modules of odm?

Best regards

matonaka, 2019-02-21:
I responded to DJI support and received an answer.

Since e-mail is coming in Japanese, I will post the translated text in English on the translation site as it is.

We become indebted to.
It is DJI JAPAN Technical Support.

I sincerely apologize for taking time to answer.

I checked with the department in charge of headquarters and now it is within the EXIF ​​information due to a product firmware bug
An abnormality occurs in the altitude calculation to be recorded.

* This problem occurs only in calculation of the value entered in the EXIF ​​information,
The altitude which is recognized by the altitude display on the application and the barometer of the aircraft is normal.

According to the specifications of our products including Mavic 2, errors are always generated at the altitude in the Exif information
It will be relieved with future Mavic 2 firmware update.
At present, the release date etc. are undecided.

I apologize for any inconvenience,
We appreciate your understanding.

closed 2019-04-10 by pierotofy

pierotofy, 2019-04-10:
Images without an altitude field should now be much better supported. This includes the mavic 2 zoom dataset. Please update, test and re-open if this is not solved?

leighghunt, 2019-04-10:
> Images without an altitude field should now be much better supported. This includes the mavic 2 zoom dataset. Please update, test and re-open if this is not solved?

Seems to work with my quick test just now with latest docker images. `adv 2...` and `adv 3...` projects using same datasets of small set of 4 images - only difference between them is latest docker images. 

[image]

I'll retest with larger dataset, but looks good. Thanks for your work @pierotofy :-)

PR #968 (2019-04-10): Extract_utm_coords replacement for C++ program
I've rewrote the UtmExtractor program in Python, which will allow for the removal of a few dependencies, speed up runtime and compile time as well as making support for EXIF tags more consistent throughout. I believe this will also fix #946 since we won't have to rely on Exiv2.

TODO before merging:
 - Test south UTM zones
 - Remove old C++ module
 - Remove dependencies from configure.sh, docker files