OpenDroneMap/ODM #950: AttributeError: 'ODM_Tree' object has no attribute 'dataset_resize'

mchttt, 2019-02-19:
Downloaded latest script for live CD, built live CD, updated /code/, installed dependencies including eigen 3.2.8 and recompiled with -DODM_BUILD_SLAM=ON

Running 
`./run.sh --project-path /home/odm/images/ --video /home/odm/GOPR0367.MP4 --slam-config /home/odm/config.yaml nz`
results in 
```
.....
Saving keyframe trajectory to KeyFrameTrajectory.txt ...

trajectory saved!
[DEBUG]   running cd /home/odm/images/nz/opensfm && PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages:/code/SuperBuild/src/opensfm python /code/modules/odm_slam/src/orb_slam_to_opensfm.py /home/odm/GOPR0367.MP4 /home/odm/images/nz/opensfm/KeyFrameTrajectory.txt /home/odm/images/nz/opensfm/MapPoints.txt /home/odm/config.yaml

Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_slam.py", line 82, in process
    os.symlink(tree.opensfm + '/images', tree.dataset_resize)
AttributeError: 'ODM_Tree' object has no attribute 'dataset_resize'
```

mchttt, 2019-02-19:
config.yaml
```
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------

# Camera calibration and distortion parameters (OpenCV)

Camera.fx: 817.125547318
Camera.fy: 814.661937758
Camera.cx: 997.614462582
Camera.cy: 707.959531664

Camera.k1: -0.214510808462
Camera.k2: 0.0551712740925
Camera.p1: 0.00107011762115
Camera.p2: -0.00275262363095
Camera.k3: -0.00745892662798

# Camera frames per second 
Camera.fps: 30
#59.94

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 2000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

use_pmvs: true
```

mchttt, 2019-02-26:
It seems to me that the SLAM script was imported from other older fork without any testing.
I found several forks of ODM with dataset_resize field being used, mostly older 2 years. Can anyone point me how i can process opensfm+bundler result to load ODM_Reconstruction class ?

pierotofy, 2019-03-08:
Hey @mchttt you are correct in that the SLAM section is dated. I don't think it has been touched in at least 2 years, so I'm not too surprised that it's failing. It should be possible to tweak it and make it work however, as what it does is fairly simple, it just extracts video frames as images prior to processing. We would welcome a PR that fixes it.

mchttt, 2019-03-11:
I managed to move further ( copied few lines from scripts/run_opensfm.py) so i managed to get to smvs point

But, i'm stuck at
```
[DEBUG]   running /code/SuperBuild/src/elibs/mve/apps/makescene/makescene /home/odm/images/a1/opensfm/mve /home/odm/images/a1/smvs
MVE Makescene (built on Jan  2 2019, 19:25:17)
Info: Detected Noah bundler format.
Reading Bundler file (21 cameras, 3726 features)...
Recognized 21 original images from Noah's Bundler.
Creating output directories...
Saving bundle file...
Writing bundle (21 cameras, 3726 features): /home/odm/images/a1/smvs/synth_0.out...
Processing view view_0000.mve...
Processing view view_0001.mve...
Processing view view_0002.mve...
Processing view view_0003.mve...
Processing view view_0004.mve...
Processing view view_0005.mve...
Processing view view_0006.mve...
Processing view view_0007.mve...
Processing view view_0008.mve...
Processing view view_0009.mve...
Processing view view_0010.mve...
Processing view view_0011.mve...
Processing view view_0012.mve...
Processing view view_0013.mve...
Processing view view_0014.mve...
Processing view view_0015.mve...
Processing view view_0016.mve...
Processing view view_0017.mve...
Processing view view_0018.mve...
Processing view view_0019.mve...
Processing view view_0020.mve...

Created 21 views with 21 valid cameras.
Imported 21 undistorted images.
[DEBUG]   running /code/SuperBuild/src/elibs/smvs/app/smvsrecon -t3 -a1.0 --max-pixels=409600 -o1 --debug-lvl=1   --force /home/odm/images/a1/smvs
Shading-aware Multi-view Stereo (built on Jan  2 2019, 19:29:55)

Initializing scene with 21 views...
Initialized 21 views (max ID is 20), took 1ms.
Reading Photosynther file (21 cameras, 3726 features)...
Automatic input scale: 2
Input embedding: undist-L2
Output embedding: smvs-B2
Running view selection for 21 views...  done, took 0.269s.
Resizing input images for 21 views...  done, took 1.164s.
Starting 1/21 ID: 1 Neighbors: 3 8 2 19 15 13 
Starting 2/21 ID: 0 Neighbors: 10 17 4 9 11 5 
Starting 3/21 ID: 2 Neighbors: 19 13 3 7 1 8 
SGM took: 5.575sec
########### Scale 4 ###########
Surface Status - Valid patches: 399
Removed 399 patches due to occlusions.
### Finished iteration: 0
Number of Newton steps: 0
Avg construction time: -nanms
Avg solver time: -nanms
Floating point exception
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/smvs.py", line 86, in process
    system.run('%s %s %s' % (context.smvs_path, ' '.join(config), tree.smvs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 136

```

Any hints as for what I'm missing ?

smathermather, 2020-04-14:
SLAM is removed now.

closed 2020-04-14 by smathermather