OpenDroneMap/ODM #958: Mvs-Texturing does not handle filenames with parenthesis

pierotofy, 2019-03-08:
To reproduce:
 - Create a folder with the following image names: "1 (1).JPG", "2 (2).JPG", ...
 - Run ODM
 - Witness:

```
[DEBUG]   running /code/SuperBuild/install/bin/texrecon /datasets/test/opensfm/reconstruction.nvm /datasets/test/odm_meshing/odm_mesh.ply /datasets/test/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none       -n 65535
/code/SuperBuild/install/bin/texrecon (built on Jan 22 2019, 00:57:48)
Load and prepare mesh: 
PLY Loader: comment VTK generated PLY File
Reading PLY: 94925 verts... 189639 faces... done.
Warning: Zero-length normals detected: 2 face normals, 6 vertex normals
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 5
NVM: Number of features: 0
                                                                                                                                                                                                                                                                                                            Loading 0%...terminate called after throwing an instance of 'util::Exception'
  what():  /datasets/test/opensfm/undistorted/1: No such file or directory
Aborted (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 126, in process
    '-n {nadirWeight}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134
```

pierotofy, 2019-03-08:
This is now mitigated in NodeODM via sanitization. https://github.com/OpenDroneMap/NodeODM/commit/cbf5e400275c08575031ea2e61cce30cfed56f03

I'm not sure about the best course of action here, one trick would be to rename/sanitize the image names prior to processing, but that's a destructive operation. The NVM format looks problematic as it does not specify how to handle spaces or special characters.

```
NVM_V3

5
undistorted/1 (1).JPG.png 2222.18133619 0.15162566427582716 -0.8532092631847736 0.34701869970500343 -0.35863858298244816 -14.783014369191108 -14.340491320546832 198.3124980912381 0 0
```

ChipwizBen, 2019-04-24:
Perhaps I'm over-simplifying the problem, but what about iterating through the files at start and hard linking with something like 1.JPG, 2.JPG, then destroy the links at the end?

closed 2020-04-14 by smathermather