OpenDroneMap/ODM #964: Scale problem

mesaricardo, 2019-03-29:
Hello,

I am using webODM some months ago and I have good results (EEUU, Canada), but in a particular area (Uruguay, WGS84 UTM21S) the result wasn't well. I tried with a differents flights and all had the same result.

 I don't have GCP of this areas, but all was tested with other photogrametry sofwares and the resault was well.

Description
Installation: webODM natively
Browsers: Explorer, Edge, Mozilla, Chrome
OS: Ubuntu Server 16.04 LTS

Drone: eBee
Camera: Canon PowerShot ELPH 110 HS

The result testing diferents configuration of proccess was all similars.

This is a print screen of the results:

[image]
[image]
[image]
[image]

How can fix this?

Best regards
Ricardo

smathermather, 2019-03-29:
@pierotofy -- this isn't related to how we are handling Southern Hemisphere projection, it it?

pierotofy, 2019-03-30:
Not related to that fix. I’m not sure what’s wrong with the orthos, they look pretty good to me, considering no GCPs were used?

pierotofy, 2019-03-31:
Wonder what happens if you pass `--use-fixed-camera-params` @mesaricardo ?

mesaricardo, 2019-04-01:
Hi Piero,

all of Project haven't GCP that I run in webODM. But I runed them in other
software and the result was good.

Best regards.
Ricardo

mesaricardo, 2019-04-01:
Dear Piero,

I tried with your suggestion but the result was identical to the others.

[image: Captura.JPG]

Kind regards.
Ricardo

smathermather, 2020-04-14:
This is probably addressed by the use of Brown-Conrady camera by default.

closed 2020-04-14 by smathermather