OpenDroneMap/ODM #969: Error build with docker on Jetson TX2. IMPORTANT!!

hitesh4, 2019-04-11:
nvidia@tegra-ubuntu:~/odm/OpenDroneMap$ docker build -t my_odm_image .Sending build context to Docker daemon  196.1MB
Step 1/31 : FROM phusion/baseimage
 ---> 166cfc3f6974
Step 2/31 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> df453af95170
Step 3/31 : RUN apt-get update -y   && apt-get install -y     software-properties-common   && add-apt-repository -y ppa:ubuntugis/ppa   && add-apt-repository -y ppa:george-edison55/cmake-3.x   && apt-get update -y
 ---> Running in ca0c41f84cb0
standard_init_linux.go:207: exec user process caused "exec format error"
The command '/bin/sh -c apt-get update -y   && apt-get install -y     software-properties-common   && add-apt-repository -y ppa:ubuntugis/ppa   && add-apt-repository -y ppa:george-edison55/cmake-3.x   && apt-get update -y' returned a non-zero code: 1

This is the error I get every time. If someone could help out.

pierotofy, 2019-04-11:
Almost looks like this could be a problem with docker and not ODM per-se. Have you tried seeing if there's any reference to this error in the docker repo on github?

EmbeddedAndroid, 2019-04-11:
```
tyler$ docker run mplatform/mquery phusion/baseimage
Unable to find image 'mplatform/mquery:latest' locally
latest: Pulling from mplatform/mquery
db6020507de3: Pull complete 
f11a2bcbeb86: Pull complete 
Digest: sha256:e15189e3d6fbcee8a6ad2ef04c1ec80420ab0fdcf0d70408c0e914af80dfb107
Status: Downloaded newer image for mplatform/mquery:latest
Image: phusion/baseimage
 * Manifest List: No
 * Supports: amd64/linux
```

The base image you are using is built for amd64 not arm64, that is your problem.

hitesh4, 2019-04-14:
How do I build a new one for arm64, or get an arm64 one? I tried building natively from the source, didn't work. Received some errors regarding ecto and some more followed by it.

zhudaoruyi, 2019-04-15:
Try changing

```
FROM phusion/baseimage
```
to
```
FROM arm64v8/ubuntu:16.04
```

hitesh4, 2019-04-15:
How do I change step 1, should I pull up the arm64v8 first or would I have to change something in the code?

hitesh4, 2019-04-15:
Thanks @zhudaoruyi.
There's a new error coming up, can you please have a look at it.

############Till here everything worked fine########################

Get:76 http://ports.ubuntu.com/ubuntu-ports xenial-updates/main arm64 software-properties-common all 0.96.20.8 [9440 B]
Get:77 http://ports.ubuntu.com/ubuntu-ports xenial/main arm64 xz-utils arm64 5.1.1alpha+20120614-2ubuntu2 [75.1 kB]
Get:78 http://ports.ubuntu.com/ubuntu-ports xenial-updates/main arm64 unattended-upgrades all 0.90ubuntu0.10 [32.3 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports xenial-security/main arm64 libexpat1 arm64 2.1.0-7ubuntu0.16.04.3 [58.3 kB]
Fetched 20.3 MB in 42s (482 kB/s)
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/c/cron/cron_3.0pl1-128ubuntu2_arm64.deb  400  Bad Request

E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/m/mime-support/mime-support_3.59ubuntu1_all.deb  Hash Sum mismatch

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
The command '/bin/sh -c apt-get update -y   && apt-get install -y     software-properties-common   && add-apt-repository -y ppa:ubuntugis/ppa   && add-apt-repository -y ppa:george-edison55/cmake-3.x   && apt-get update -y' returned a non-zero code: 100

closed 2020-04-14 by smathermather