OpenDroneMap/ODM #970: [Improvement] Make PDAL DEM generation memory efficient

pierotofy, 2019-04-11:
Current proof of concept works as follows:

 - Calculate point cloud extent via `pdal info --summary <pc file>`
 - Based on target cell resolution, estimate total raster size
 - If raster size exceeds a threshold, split extent into parts
 - Process `pdal` commands via `pdal pipeline -i /tmp/tmpdW_nxT.json --writers.gdal.bounds="([576705.83935, 576767.5348], [5188105.558, 5188159.2135])"` passing the proper bounds and creating the proper tiles.
 - This process is repeated for each radius
 - Use `gdalbuildvrt` to combine rasters: `gdalbuildvrt merged.vrt 1_1.tif 1_2.tif ...`
 - Create .tiff via `gdal_translate merged.vrt merged.tif`
 - Apply nearest neighbor interpolation using current numpy code
 - Success.

pierotofy, 2019-04-12:
Done!

closed 2019-04-12 by pierotofy

smathermather, 2019-04-12:
Lol. I was about to comment on the outline. :)

Great!