OpenDroneMap/ODM #977: Core dump / Error 132 during smvs

ichfred, 2019-04-23:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Natively with the webODM instalation Script on  Ubuntu 16.04.6 LTS (GNU/Linux 4.4.0-143-generic x86_64)

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 66 on Windows 10 

### What is the problem?

Once again I have a 132 Error. This time on a different step and on a different hardware then posted in a previous case: https://github.com/OpenDroneMap/ODM/issues/859

The server CPUs ( 2x AMD Opteron(tm) Processor 6172 12 Core) is not supporting SSSE3. Running the aukerman sample files. I installed WebODM with the ServerScript nativly on a fresh Ubuntu 16 installation. So it should be a compiled version as it is written in the FAQ? The HP Proliant Server has 63GB of RAM. So this should also don't be a problem.

Tested with different WebODM Settings: "High Resolution" Options: min-num-features: 18000, orthophoto-resolution: 2.0, dem-resolution: 2.0, dsm: true, depthmap-resolution: 1000 or the "Fores"t presets

`[INFO] Running ODM Load Dataset Cell
[DEBUG] Loading dataset from: /www/data/6869d432-05d7-4ca0-92b0-d5bdb7f86404/images
[DEBUG] /www/data/6869d432-05d7-4ca0-92b0-d5bdb7f86404/images
[...] opensfm extract_metadata
[...] opensfm detect_features
[...] opensfm match_features
[...] opensfm create_traks
[...] opensfm reconstruct
[...] opensfm export_visualsfm
[...] opensfm undistort
[...] opensfm export_bundler
[...] opensfm export_geocoords
[...] makescene
Processing view view_0019.mve...
Created 77 views with 77 valid cameras.
Imported 77 undistorted images.
[DEBUG] running /code/SuperBuild/src/elibs/smvs/app/smvsrecon -t24 -a1.0 --max-pixels=1000000 -o1 --debug-lvl=0 /www/data/6869d432-05d7-4ca0-92b0-d5bdb7f86404/smvs
Shading-aware Multi-view Stereo (built on Mar 31 2019, 20:35:53)

Initializing scene with 77 views...
Initialized 77 views (max ID is 76), took 10ms.
Reading Photosynther file (77 cameras, 156068 features)...
Illegal instruction (core dumped)
Traceback (most recent call last):
File "/code/run.py", line 47, in
plasm.execute(niter=1)
File "/code/scripts/smvs.py", line 86, in process
system.run('%s %s %s' % (context.smvs_path, ' '.join(config), tree.smvs))
File "/code/opendm/system.py", line 34, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132`

Is it possible to override the CPU features or something like this? Thanks for your support.

The topic is also (with less acceptance) discussed here: https://community.opendronemap.org/t/exit-on-code-132/766/8

pierotofy, 2019-04-23:
Hey @ichfred have you tried to rebuild the docker image on the target machine (although I suspect you won't be able to compile it because of the lack of those instructions).

`docker build -t opendronemap/odm .`

ichfred, 2019-04-23:
Hmm.. It isn't a docker installation yet. But I can give it a try.

pierotofy, 2019-04-23:
Ah, never mind. Missed the part where you mentioned this is a native install.

Pass the —use-opensfm-dense parameter, that should skip smvs and perhaps you’ll be able to run it.

ichfred, 2019-05-17:
I found the issue. There was a problem with the coord written in the EXIF by the PPK programm. There is no problem with the latest ODM/WebODM anymore.

Thanks.
Fred

closed 2019-05-17 by ichfred