OpenDroneMap/ODM #982: Can't signup/login to the community forum.

MatthiasSiewert, 2019-05-14:
I tried several methods including (email, github and twitter) to create an account or sign in to:
https://community.opendronemap.org/

The confirmation email never arrived on my email account (tried several email addresses) and the other methods returned an authentication issue:

' Sorry, there was an error authorizing your account. Perhaps you did not approve authorization? '

ismailsunni, 2019-05-14:
I have the same problem

smathermather, 2019-05-14:
Thanks. Looking into it.

smathermather, 2019-05-14:
Try now @ismailsunni and @MatthiasSiewert.

ismailsunni, 2019-05-14:
@smathermather I still got errors for Github, twitter, and Google.

smathermather, 2019-05-14:
Can you log in via email? I will look into the others.

MatthiasSiewert, 2019-05-14:
Thanks for looking into it. I can not log in via email. I didn't have an account before and haven't recieve a confirmation email yet.

luipir, 2019-05-15:
Hey... I've the same problem. can't login via basic auth or oauth too

smathermather, 2019-05-15:
Thanks for the updates: I can't replicate this with basic auth. Can you give me more details? I signed up with a new email account a few minutes ago, it sent the activation link, and I am now logged in.

MatthiasSiewert, 2019-05-15:
When I try with google, I get the following message, maybe that helps:

400. That’s an error.

Error: redirect_uri_mismatch

The redirect URI in the request, https://community.opendronemap.org/auth/google_oauth2/callback, does not match the ones authorized for the OAuth client. To update the authorized redirect URIs, visit: https://console.developers.google.com/apis/credentials/oauthclient/1081421517513-vjr8tsgnak98s12bv036fud0rrjnh9vc.apps.googleusercontent.com?project=1081421517513

luipir, 2019-05-15:
I'm able to sign-in with mail link, and with basic auth after a password reset.
Not yet able to sign-in with oauth (tested with google and gtihub).
github error: "Sorry, there was an error authorizing your account. Perhaps you did not approve authorization?" but in my github config I do not have any restriction to ODM
google error: as notifed by @MatthiasSiewert

MatthiasSiewert, 2019-05-15:
Still none of the options works for me. Maybe I am a robot....?!?

smathermather, 2019-05-16:
Probably not a robot. Give me a few days, I need a block of time to troubleshoot.

smathermather, 2019-05-23:
Piero got this working again. I need to finish up with twitter login but the rest should be working as of a few days ago.

pierotofy, 2019-09-26:
This should be fixed, please re-open if the problem persists.

closed 2019-09-26 by pierotofy

mj-saunders, 2020-11-08:
I am also not receiving the activation link email.

pierotofy, 2020-11-08:
Did you check the spam folder? What's your username on the forum?

mj-saunders, 2020-11-08:
@pierotofy Yes, nothing at all. Username 'berilac'

pierotofy, 2020-11-09:
I've manually activated your account :+1:

mj-saunders, 2020-11-09:
@pierotofy Thank you! :)

camcarto, 2023-07-01:
hi, ive recently started using WebODM. I am new and would like to join the forum but I am unable to create an account. Ive tried two email accounts and google sign up each time "New registrations are not allowed from your IP address (maximum limit reached). Contact a staff member." Is it not possible to sign up from South Africa?

Saijin-Naib, 2023-07-05:
Hey, please reach out to hello@uav4geo.com and I'll work with you to get your account sorted.

camcarto, 2023-07-05:
> Hey, please reach out to [hello@uav4geo.com](mailto:hello@uav4geo.com) and I'll work with you to get your account sorted.

Thank you @Saijin-Naib! i have sent 👍