OpenDroneMap/ODM #988: [Feature Request ] Faster Merging Merge step of Split Merge

smathermather, 2019-05-28:
`gdalwarp` is used for the merge step in split-merge approaches. gdalwarp is quite a fast way to merge the data, unless blending is used, at which point it becomes painfully slow. It would be nice to either replace gdalwarp at this step with an optimized solution or optimize gdalwarp and contribute upstream.

pierotofy, 2019-05-28:
I think part of the bottleneck is the code in https://github.com/OSGeo/gdal/blob/d5f6bb89b0e0db0a06489419050d432e8f58ff47/gdal/alg/gdalcutline.cpp#L57 so perhaps work could simply go toward optimizing that routine in GDAL.

pierotofy, 2019-05-28:
Also, I think it's safe to say that lanczos interpolation is probably overkill, we should switch to bilinear or even near since most orthophotos will end up having the same resolution (we're just stitching them together, we're not scaling them, for the most part). This alone should give a 2-3x speed up.

smathermather, 2019-05-28:
Ya, lanczos is overkill for most non-quantitative applications. It still makes some sense for elevation models, as it keeps the features from moving too much.

pierotofy, 2019-05-31:
`bilinear` interpolation is now used in #992

pierotofy, 2020-03-19:
This has been done some time ago. :clinking_glasses:

closed 2020-03-19 by pierotofy

smathermather, 2020-03-19:
And boy is it much faster and less memory intensive. I've never seen PDAL work so hard.