OpenDroneMap/ODM #991: [Feature Request ] Parallelize odm_filterpoints

smathermather, 2019-05-28:
Parallelize odm_filterpoints to reduce the bottleneck at this stage in the process.

pierotofy, 2019-06-22:
This can be done when/if this gets done https://github.com/PDAL/PDAL/issues/2570

kornyhiv, 2019-09-06:
Looks like it has been completed:
https://github.com/PDAL/PDAL/issues/2570

pierotofy, 2019-09-07:
Some work has been started here https://github.com/pierotofy/OpenDroneMap/tree/parallel_filterpoints but it might be a while before I can finish it. @kornyhiv would you be interested in it?

pierotofy, 2020-06-03:
We can now raise funds to get this done using quadratic funding! https://fund.webodm.org/#/fund/1

pierotofy, 2020-07-11:
See #1132 for updates. This is now done! Please test and see if performance improves. Thanks to everyone who contributed!

closed 2020-07-11 by pierotofy