OpenDroneMap/ODM #998: Bad Reconstructions Cause PDAL to hang forever

pierotofy, 2019-06-21:
Sometimes if a reconstruction is degenerate it may span a very large area and cause the DEM generation to attempt to generate hundred of tiles (which will never complete).

```
[INFO]    Generating /var/www/data/2442146d-4287-4c96-ad2e-0737c36a703f/odm_meshing/tmp/mesh_dsm_r0.07228855395144211_x0_y1.tif (max, radius: 0.07228855395144211, resolution: 0.025557863350617153)
[DEBUG]   running pdal pipeline -i /tmp/tmpOtDfQ6.json > /dev/null 2>&1
#> !!
[INFO]    DEM resolution is (9991166, 1581849), max tile size is 2048, will split DEM generation into 484 tiles
```

This should be caught (how?) and process stopped with failure.

smathermather, 2020-04-14:
Is this still a bug?

pierotofy, 2020-04-14:
I think so, although it might be mitigated by point cloud filtering. I'd leave it open.

pierotofy, 2021-04-05:
I haven't see this one in a while. Closing.

closed 2021-04-05 by pierotofy

smathermather, 2021-04-05:
I haven't seen it even with some pretty silly point clouds.