OpenDroneMap/ODM #1002: Running mve stage: Error loading: *.JPG (skipping)

KeoChi, 2019-06-27:
****************************************

### How did you install OpenDroneMap? (Docker, natively, ...)?

   natively

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

  Ubuntu 16.04(whatismybrowser.com/w/EY3SEYA)

### What is the problem?

   Hello! When I run the newest code, I get the error:
   ```
Running mve stage
[DEBUG]   running /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/makescene/makescene /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/dataset/submodels/submodel_0000/opensfm/mve /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/dataset/submodels/submodel_0000/mve
MVE Makescene (built on Jun 27 2019, 16:44:49)
Info: Detected Noah bundler format.
Reading Bundler file (87 cameras, 89159 features)...
Recognized 87 original images from Noah's Bundler.
Creating output directories...
Saving bundle file...
Writing bundle (87 cameras, 89159 features): /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/dataset/submodels/submodel_0000/mve/synth_0.out...
Processing view view_0000.mve...
Processing view view_0022.mve...
Processing view view_0044.mve...
Processing view view_0066.mve...
Error loading: /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/dataset/submodels/submodel_0000/images/0145.JPG (skipping)
Processing view view_0023.mve...
Error loading: /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/dataset/submodels/submodel_0000/images/0146.JPG (skipping)
Processing view view_0001.mve...
Error loading: /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/dataset/submodels/submodel_0000/images/0102.JPG (skipping)
```
  The mve can't read the soft link image.

pierotofy, 2019-06-27:
Hey @KeoChi :hand: does this happen with all images or just some?

KeoChi, 2019-06-28:
Hello,@pierotofy ! I tested another set of data sets and I didn't see such an error. I will test more dataset later.
  In addition, there are other errors.
  ```
[DEBUG]   running /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon -s3 --progress=silent --local-neighbors=2 /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/DJI_2000_RTK/submodels/submodel_0000/mve
MVE Depth Map Reconstruction (built on Jun 27 2019, 16:44:46)
Initializing scene with 62 views...
Initialized 62 views (max ID is 61), took 6ms.
Reading Photosynther file (62 cameras, 58999 features)...
Reconstructing all views...
*** Error in `/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon': corrupted double-linked list: 0x0000000000c72020 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fcb2ed427e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x82970)[0x7fcb2ed4d970]
/lib/x86_64-linux-gnu/libc.so.6(__libc_malloc+0x54)[0x7fcb2ed4f184]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_Znwm+0x18)[0x7fcb2fa92498]
/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon[0x41f601]
/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon[0x41f180]
/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon[0x411860]
/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon[0x41445d]
/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon[0x40ae70]
/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon[0x40c237]
/usr/lib/x86_64-linux-gnu/libgomp.so.1(GOMP_parallel+0x3f)[0x7fcb2f4d6d9f]
/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon[0x4083aa]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7fcb2eceb830]
/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon[0x4097b9]
======= Memory map: ========
00400000-00490000 r-xp 00000000 08:03 6558934                            /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon
0068f000-00691000 r--p 0008f000 08:03 6558934                            /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon
00691000-00692000 rw-p 00091000 08:03 6558934                            /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon
00692000-00694000 rw-p 00000000 00:00 0 
00c47000-25df8000 rw-p 00000000 00:00 0                                  [heap]

[WARNING] Caught error code, retrying attempt #2
[DEBUG]   running /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/SuperBuild/src/elibs/mve/apps/dmrecon/dmrecon -s3 --progress=silent --local-neighbors=2 /home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/DJI_2000_RTK/submodels/submodel_0000/mve
MVE Depth Map Reconstruction (built on Jun 27 2019, 16:44:46)
Initializing scene with 62 views...
Initialized 62 views (max ID is 61), took 1480ms.
Reading Photosynther file (62 cameras, 58999 features)...
Reconstructing all views...
Reconstruction took 298267ms.
Saving views back to disc...
Saving views to MVE files... done
```
  The first time it ran the dmrecon of mve, it made a mistake. And then "retrying attempt #2", it run successfully. Why is there such a situation?

   Another error:
  ```
 [INFO]    Starting smoothing...
[INFO]    Smoothing iteration 1
Traceback (most recent call last):
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/run.py", line 56, in <module>
    app.execute()
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/stages/odm_app.py", line 91, in execute
    self.first_stage.run()
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/opendm/types.py", line 376, in run
    self.next_stage.run(outputs)
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/opendm/types.py", line 376, in run
    self.next_stage.run(outputs)
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/opendm/types.py", line 376, in run
    self.next_stage.run(outputs)
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/opendm/types.py", line 376, in run
    self.next_stage.run(outputs)
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/opendm/types.py", line 376, in run
    self.next_stage.run(outputs)
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/opendm/types.py", line 376, in run
    self.next_stage.run(outputs)
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/opendm/types.py", line 357, in run
    self.process(self.args, outputs)
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/stages/odm_meshing.py", line 72, in process
    smooth_dsm=not args.fast_orthophoto)
[... 5 lines cut ...]
    arr = ndimage.median_filter(arr, size=5, output=dtype)
  File "/usr/lib/python2.7/dist-packages/scipy/ndimage/filters.py", line 1069, in median_filter
    origin, 'median')
[... cut ...]
    self.process(self.args, outputs)
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/stages/splitmerge.py", line 154, in process
    system.run(" ".join(map(quote, argv)), env_vars=os.environ.copy())
  File "/home/keo/nnd_storage/keo_second_workspace/OpenDroneMap_new/ODM/opendm/system.py", line 76, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```
  I don't know how to avoid this situation at the moment. 
  Looking forward to your reply. Thank you!

KeoChi, 2019-06-28:
I have already solved the last error. It related to "scipy".  I rm the older "scipy" and *scipy*.egg-info in /usr/local/lib/python2.7/dist-packages. And then "sudo pip install -U scipy==1.2.1".

pierotofy, 2019-06-28:
dmrecon has a race condition in their code (which gets triggered about 1 out of 10 times). We haven't been able to fix it, but we'd welcome contributions to fix it upstream. For the time being, we just retry it until it works.

KeoChi, 2019-06-29:
Excuse me. @pierotofy  I have another problem.  When running this step in [cropper.py](https://github.com/OpenDroneMap/ODM/blob/master/opendm/cropper.py):
`run('pdal info --boundary --filters.hexbin.edge_size=1 --filters.hexbin.threshold=0 {0} > {1}'.format(decimated_pointcloud_path,  boundary_file_path))`
  There is an error. Like:
```
(pdal info filters.hexbin Error) GDAL failure (1) OGR_G_RemoveGeometry() not supported on polygons yet.
(pdal info filters.hexbin Error) GDAL failure (1) OGR_G_RemoveGeometry() not supported on polygons yet.
(pdal info filters.hexbin Error) GDAL failure (1) OGR_G_RemoveGeometry() not supported on polygons yet.
(pdal info filters.hexbin Error) GDAL failure (1) OGR_G_RemoveGeometry() not supported on polygons yet.
(pdal info filters.hexbin Error) GDAL failure (1) OGR_G_RemoveGeometry() not supported on polygons yet.
```
   Although this error occurs, the file "odm_georeferenced_model.boundary.json" can still be generated.　The follow-up process works normally. I don't understand why this error occurs.

pierotofy, 2019-06-29:
It's a warning and can be ignored. It will be fixed when we upgrade PDAL.

closed 2019-06-29 by pierotofy