OpenDroneMap/ODM #1005: [Feature Request] Add SMVS as an option

fflores98chile, 2019-07-02:
It would be great to have SMVS back as an option to MVE, so you can select one or the other depending on your scene and objective. 
We have been working with some complex forest scenes and SMVS is doing a better job on creating the dense point cloud for areas with canopy closure and taller trees. In areas in which we have small individual trees, MVE seems to be better (SMVS is smoothing out some small isolated trees).
We have also experience a significant increase in processing time at high depthmap-resolution with MVE (for large datasets ~4.000 images, increasing from 3 to 8 days).
Thanks,

Francisco

pierotofy, 2019-07-02:
Hey @fflores98chile :hand: SMVS is certainly a bit better at filling certain gaps in the point cloud (at the expense of smoothed, less geometrically precise results).

A few questions:
 - What happens if you lower `--mve-confidence`?
 - Are you sure the increase in runtime is due to MVE vs. SMVS? Could you post the `benchmark.txt` file obtained from running ODM with the `--time` option from both versions?

fflores98chile, 2019-07-03:
Hi Piero,
I ran with mve-confidence at 0.01 and it was still incomplete (see images below).
In term of the benchmark run-time, my comparison was between two runs I did with the same dataset  in WebODM Lightning, before and after the change replacing SMVS. I cannot run those larger datasets in my server.
Francisco

[image]
[image]

fflores98chile, 2019-07-03:
Just to clarify, the top image is MVE at 0.01 -mve-confidence, and the bottom is SMVS.

Francisco

smathermather, 2019-07-15:
@fflores98chile -- have you tried [NodeMICMAC](https://github.com/dronemapper-io/NodeMICMAC)? I'm curious how it would do with this dataset.

fflores98chile, 2019-07-22:
Hi,

Yes, I tried in NodeMICMAC and the reconstruction failed. I also tried another dataset with simpler structure (tree orchard with clearly defined trees) and NodeMICMAC completed the reconstruction. I'm still trying to have a better handle of NodeMICMAC, but with what I know so far, I have not been able to complete the reconstruction of the complex forest dataset.

I also did a test with the same complex forest dataset on Pix4D and the reconstruction was also incomplete (a lot of gaps in the point cloud), similar to MVE, so the best results so far were obtained with SMVS.

smathermather, 2019-08-21:
Did you post the failing dataset to NodeMICMAC github issues? The failure to reconstruct is probably solvable.

closed 2020-04-14 by smathermather

pierotofy, 2020-04-14:
Yep, not enough bandwidth to implement this. We'd welcome a PR to add support for SMVS.

smathermather, 2020-04-14:
👍