OpenDroneMap/ODM #1008: MVS Texturing: Child returned 137

merkato, 2019-07-16:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker install on Lubuntu 19.04, with 4 GB RAM, nodeodm image.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 75 on Linux

### What is the problem?

This was processing of 37 photos test dataset, Everything is going well until MVS Texturing step

Running...
done. (Took 12.558s)
2516 texture patches.
Running global seam leveling:
Create matrices for optimization...
done.
Lhs dimensionality: 110658 x 110658
Calculating adjustments:
Color channel 1: CG took 93 iterations. Residual is 9.86748e-05
Color channel 0: CG took 105 iterations. Residual is 9.71017e-05
Color channel 2: CG took 105 iterations. Residual is 9.74425e-05
Took 0.499 seconds
Adjusting texture patches 100%... done. (Took 3.264s)
Running local seam leveling:
Killed
Traceback (most recent call last):
File "/code/run.py", line 56, in 
app.execute()
File "/code/stages/odm_app.py", line 93, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 376, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 376, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 376, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 376, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 376, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 376, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 357, in run
self.process(self.args, outputs)
File "/code/stages/mvstex.py", line 97, in process
'-n {nadirWeight}'.format(**kwargs))
File "/code/opendm/system.py", line 76, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Expected behaviour? Probably orthophoto would be nice ;)

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Reproducible with max-concurrency set to 1 or 2, use-opensfm-dense, on every dataset I'm trying. 

Any chance that building docker image ourself on this machine will resolve this issue?

pierotofy, 2019-07-17:
You are running out of memory. How does your 3D mesh / point cloud look like? You probably have a bad reconstruction and the size of the mesh ends up being too big.

pierotofy, 2019-07-18:
Could we move the conversation over to https://community.opendronemap.org ? Thank you.

closed 2019-07-18 by pierotofy