OpenDroneMap/ODM #1009: Brown–Conrady model fails in OpenSFM

gohai, 2019-07-18:
### How did you install OpenDroneMap?

Docker

### What's your browser and operating system?

Amazon Linux (CentOS-based)

### What is the problem?

The OpenSFM included in ODM fails for me when I try to use a `cameras.json` with a Brown-Conrady type correction.

The relevant output I see from running ODM:

```
  File "/code/SuperBuild/src/opensfm/bin/export_bundler", line 48, in <module>^M
    io.export_bundler(images, reconstructions, track_graph, bundle_file_path, list_file_path)^M
  File "/code/SuperBuild/src/opensfm/opensfm/io.py", line 703, in export_bundler^M
    focal = camera.focal * scale^M
AttributeError: 'BrownPerspectiveCamera' object has no attribute 'focal'^M
```

This is the `cameras.json` file I used:

```
{
    "dji fc2204 4000 3000 perspective 0.6666": {
        "projection_type": "brown", 
        "width": 4000, 
        "height": 3000, 
        "c_x": 0.492639, 
        "c_x_prior": 0.492639, 
        "c_y": 0.503773, 
        "c_y_prior": 0.503773, 
        "k1": -0.040661, 
        "k1_prior": -0.040661, 
        "k2": 0.104878, 
        "k2_prior": 0.104878, 
        "k3": -0.095506, 
        "k3_prior": -0.095506, 
        "p1": 0.000122, 
        "p1_prior": 0.000122, 
        "p2": 0.000276, 
        "p2_prior": 0.000276, 
        "focal": 0.721327, 
        "focal_prior": 0.721327, 
        "focal_x": 0.721327, 
        "focal_x_prior": 0.721327, 
        "focal_y": 0.721327, 
        "focal_y_prior": 0.721327
    }
}
```

pierotofy, 2019-07-18:
Hey @gohai let's move the conversation over to https://community.opendronemap.org as this looks like a question and not a bug? Thank you! :pray:

gohai, 2019-07-18:
Hi @pierotofy - I _do_ believe this is a bug. After all I am running an unmodified `ODM` install, which just fails deep inside OpenSfM when I use the `brown`-type projection type (it works when I use the `perspective` one).

I also found another issue filled against OpenSfM where their code didn't handle `brown`-type projections elsewhere: https://github.com/mapillary/OpenSfM/issues/259

But since you're the main author of this great piece of software, I'd be a fool to argue - and I'll dully go post this on the community page! Have a good day.

pierotofy, 2019-07-18:
Hey @gohai, I apologize, I think you are right that this in fact might be a bug. I wonder if an issue should be opened on the OpenSfM project however, as it looks related to it more than ODM.

gohai, 2019-07-18:
I filled this as https://github.com/mapillary/OpenSfM/issues/447

YanNoun, 2019-08-20:
Hi,

We fixed it here : https://github.com/mapillary/OpenSfM/issues/447#issuecomment-515883967

Yann

pierotofy, 2019-08-20:
Yep, thanks @YanNoun ! 

The OpenSfM changes are available in #1015 but haven't been merged into master yet.

pierotofy, 2019-09-09:
Merged and fixed.

closed 2019-09-09 by pierotofy