OpenDroneMap/ODM #1012: Warn users that if a dataset has no GPS info, but split-merge is enabled

pierotofy, 2019-07-30:
split-merge either needs GPS info or a `image_groups.txt` file to split a dataset.

Otherwise it cannot work. The program should help users understand that instead of failing with :

```
OpenCV Error: Assertion failed (data0.dims <= 2 && type == CV_32F && K > 0) in kmeans, file /code/SuperBuild/src/opencv/modules/core/src/matrix.cpp, line 2701
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/create_submodels.py", line 30, in run
    self._cluster_images(meta_data, data.config['submodel_size'])
  File "/code/SuperBuild/src/opensfm/opensfm/commands/create_submodels.py", line 100, in _cluster_images
    labels, centers = tools.kmeans(positions, K)[1:]
  File "/code/SuperBuild/src/opensfm/opensfm/large/tools.py", line 34, in kmeans
    return cv2.kmeans(samples, nclusters, criteria, attempts, flags)
cv2.error: /code/SuperBuild/src/opencv/modules/core/src/matrix.cpp:2701: error: (-215) data0.dims <= 2 && type == CV_32F && K > 0 in function kmeans
```

theoway, 2022-08-04:
I'd love to make a PR. Here's how I'm thinking of solving it. If you approve it, I'll make the changes then :)

- Add a member variable `self.has_geotagged_photos` and a function `has_gps()` that indicates if GPS info is available.
https://github.com/OpenDroneMap/ODM/blob/6fee4be2b655bd5bf288aaa52759d4d1a6ecc940/opendm/types.py#L94-L95

- Perform a check here:
https://github.com/OpenDroneMap/ODM/blob/6fee4be2b655bd5bf288aaa52759d4d1a6ecc940/stages/splitmerge.py#L32-L35

pierotofy, 2022-08-04:
Sounds like a good approach!

theoway, 2022-08-04:
> Sounds like a good approach!

Could you tell me what would be better? Exiting the process or just logging a warning?

pierotofy, 2022-08-04:
I would print a warning and try to process the dataset without split-merge.

PR #1521 (2022-08-05): Added a warning before split-merge to ensure GPS/`image_groups.txt` info is present.
Fixes #1012 

This adds a check before initiating `split-merge` to see if either GPS info in photos or `image_groups.txt` is available. If not, it skips `split-merge` and proceeds with a warning.
closed 2022-08-05 by pierotofy