OpenDroneMap/ODM #1016: Split-merge model bounds issues

x-ancin, 2019-08-12:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker in CentOS linux

### What's your browser and operating system? 

CentOS linux 7 installed on a server

### What is the problem?

I have been processing large datasets (approximately 3000 images) using the split-merge function

I use the following command:
docker run -it --rm -v /images:/datasets/code opendronemap/odm --project-path /datasets --split 150 --split-overlap 10 --ignore-gsd --orthophoto-resolution 0.3 --dem-resolution 0.3 --dsm --merge pointcloud

The submodels are correctly processed, and I can get the orthophotos from there, but they are not clipped to the proper extent. The process finishes as a failed process with the following error:

[INFO] running pdal translate “ept:///datasets/code/entwine_pointcloud” "/datasets/code/odm_georeferencing/odm_georeferenced_model.laz"e[0m
(pdal translate readers.ept Warning) 492221687 will be downloaded
e[39m[INFO] Merging all crop bounds: [’/datasets/code/submodels/submodel_0000/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0001/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0002/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0003/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0004/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0005/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0006/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0007/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0008/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0009/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0010/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0011/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0012/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0013/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0014/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0015/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0016/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0017/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0018/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’, ‘/datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.bounds.gpkg’]e[0m
Traceback (most recent call last):
File “/code/run.py”, line 56, in 
app.execute()
File “/code/stages/odm_app.py”, line 93, in execute
self.first_stage.run()
File “/code/opendm/types.py”, line 376, in run
self.next_stage.run(outputs)
File “/code/opendm/types.py”, line 376, in run
self.next_stage.run(outputs)
File “/code/opendm/types.py”, line 357, in run
self.process(self.args, outputs)
File “/code/stages/splitmerge.py”, line 206, in process
Cropper.merge_bounds(all_bounds, merged_bounds_file, 0)
File “/code/opendm/cropper.py”, line 82, in merge_bounds
layer = ds.GetLayer()
AttributeError: ‘NoneType’ object has no attribute ‘GetLayer’

It seems like there is some problem with the .gpkg files, as seen in the output for every submodel:

[INFO] running gdalwarp -cutline /datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.bounds.gpkg -crop_to_cutline -co BLOCKXSIZE=512 -co COMPRESS=LZW -co TILED=YES -co NUM_THREADS=8 -co BLOCKYSIZE=512 /datasets/code/submodels/submodel_0019/odm_dem/dsm.original.tif /datasets/code/submodels/submodel_0019/odm_dem/dsm.tif --config GDAL_CACHEMAX 46.55%e[0m
ERROR 1: Cannot open /datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.bounds.gpkg.
e[93m[WARNING] Something went wrong while cropping: Child returned 1e[0m
e[39m[INFO] Cropping /datasets/code/submodels/submodel_0019/odm_dem/dsm.unfilled.tife[0m
e[39m[INFO] running gdalwarp -cutline /datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.bounds.gpkg -crop_to_cutline -co BLOCKXSIZE=512 -co COMPRESS=LZW -co TILED=YES -co NUM_THREADS=8 -co BLOCKYSIZE=512 /datasets/code/submodels/submodel_0019/odm_dem/dsm.unfilled.original.tif /datasets/code/submodels/submodel_0019/odm_dem/dsm.unfilled.tif --config GDAL_CACHEMAX 46.55%e[0m
ERROR 1: Cannot open /datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.bounds.gpkg.
e[93m[WARNING] Something went wrong while cropping: Child returned 1e[0m

Which seems to come from some issue with GDAL itself:

[INFO] Calculating cropping area and generating bounds shapefile from point cloude[0m
e[39m[INFO] running pdal translate -i “/datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.laz” -o “/datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.decimated.las” decimation --filters.decimation.step=90 e[0m
e[39m[INFO] running pdal info --boundary --filters.hexbin.edge_size=1 --filters.hexbin.threshold=0 /datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.decimated.las > /datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.boundary.jsone[0m
(pdal info filters.hexbin Error) GDAL failure (1) OGR_G_RemoveGeometry() not supported on polygons yet.
e[39m[INFO] running pdal info --summary /datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.laz > /datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.summary.jsone[0m
e[39m[INFO] running ogr2ogr -overwrite -f GPKG -a_srs “+proj=utm +zone=11 +datum=WGS84 +units=m +no_defs” /datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.bounds.gpkg /datasets/code/submodels/submodel_0019/odm_georeferencing/odm_georeferenced_model.bounds.geojsone[0m
ERROR 1: sqlite3_exec(COMMIT) failed: database is locked
ERROR 1: sqlite3_exec(PRAGMA synchronous = OFF) failed: Safety level may not be changed inside a transaction
ERROR 1: sqlite3_exec(BEGIN) failed: cannot start a transaction within a transaction
ERROR 1: Transaction not established

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Cut the bounds of the orthophotos on each submodel processing and create a merged orthophoto

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

It happens with any image set I upload, and any split/split overlap combinations

OKaluza, 2020-05-27:
I think I'm having the same issue. Is your data volume a cifs mount?

The odm_georeferenced_model.bounds.gpkg files were all zero bytes and I tracked it down to an earlier error running `ogr2ogr` that produces the same sqlite3 error you posted.

In my case, I have a cifs mount for the data and it seems the error is due to sqlite3 being incompatible with cifs https://github.com/r-spatial/sf/issues/628

I am currently trying to run my job again with a patched cropper.py that outputs the .gpkg file to a temporary directory and then moves it to the data dir

pierotofy, 2023-11-05:
Quick update on this one; please try with the latest version, we've made a lot of improvements since this report has been opened.

If the issue persists, please share a datasets and reproduction instructions? 🙏

closed 2023-11-05 by pierotofy