OpenDroneMap/ODM #1020: Clarify error when running out of RAM due to large dimensions of generated orthophoto

skorasaurus, 2019-08-30:
****************************************

### How did you install OpenDroneMap? (Docker, natively, ...)?

[Type answer here]

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

I received an error that states "Try to reduce the -resolution parameter or add more RAM." 

This error message is found specifically in at https://github.com/OpenDroneMap/ODM/blob/d80d0b2992c96d4a336319f46e76b844dfa33e84/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp#L40

[The list of command line arguments](https://docs.opendronemap.org/using.html#command-line-arguments) does not have an there is no argument for -resolution  

But from parsing that OdmOrthoPhoto.cpp file and my limited knowledge of C and the immediate lines hat proceed https://github.com/OpenDroneMap/ODM/blob/d80d0b2992c96d4a336319f46e76b844dfa33e84/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp#L40  should the error instead refer to 
--orthophoto-resolution and direct the user to increase the argument (which in turn decrease the dimensions of the photo).

pierotofy, 2019-08-31:
Good catch, yes this is a misleading error message and should probably be changed; the reason it refers to `-resolution` is because the `odm_orthophoto` program has a `-resolution` parameter, but the ODM pipeline uses `--orthophoto-resolution` (see https://github.com/OpenDroneMap/ODM/blob/master/stages/odm_orthophoto.py#L64).

I think we could simply change the error message to reference `--orthophoto-resolution`. Would you be interested in contributing a PR for this?

skorasaurus, 2019-09-01:
I'm happy to make a PR. But before I do, I want to make sure that I understand correctly: 

For example, if the user hits this error because they ran out of memory, _I believe that they should increase the orthophoto-resolution_; which is cm/pixel. This is based from the https://docs.opendronemap.org/using.html#creating-high-quality-orthophotos recommendation that says to decrease the value for for higher resolution and higher quality photor

I've verified this by running odm twice on the same image set on the same server:
 once without specifying the orthophoto-resolution and [it failed to finish](https://gist.github.com/skorasaurus/7901e99c9d3628ab0fef08a5771edcce#file-2019-09-01-take8-log-L411) and tried to produce an image of 40992x57124 

the second time, I applied `--orthophoto-resolution 20` it successfully finished and produced a smaller image OF 10380x14318

This contrasts the [resolution argument](https://github.com/OpenDroneMap/ODM/blob/d80d0b2992c96d4a336319f46e76b844dfa33e84/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp#L297) is the inverse, which is pixels/meter. 

I propose then, to change the message in to `Try to increase the --orthophoto-resolution to a larger integer`. Before I do and submit the PR, could you confirm if I am correct?

pierotofy, 2019-09-01:
You are correct :+1:

commit 9ea78ba6 (2019-09-02): clarify error message when running out of ram for generating orthphoto, fix #1020
commit d814eef4 (2019-09-02): clarify error msg when when running out of ram for generating orthphoto, fix #1020
closed 2019-09-02 by pierotofy