OpenDroneMap/ODM #1022: Child Return 137 during "Generating texture patches: Running .... Killed

pjmaechling, 2019-09-03:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Using docker image, using default command line invocation of docker.
Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

whatismybrowser.com/w/C6LZ8FJ

### What is the problem?

With a collection of 270 Phantom 4 images collected with dronedeploy,
I run this command:

 docker run -ti --rm -v /Users/maechlin/mydocker/odm:/datasets/code opendronemap/odm --project-path /datasets

And  I get this error

Program exits during texture mapping:
NFO]    Writing MVS Textured file in: /datasets/code/odm_texturing/odm_textured_model.obj
[INFO]    Removing old tmp directory /datasets/code/odm_texturing/tmp
[INFO]    running /code/SuperBuild/install/bin/texrecon /datasets/code/opensfm/reconstruction.nvm /datasets/code/odm_meshing/odm_mesh.ply /datasets/code/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none       -n 65535
/code/SuperBuild/install/bin/texrecon (built on Aug 13 2019, 21:43:57)
Load and prepare mesh: 
PLY Loader: comment VTK generated PLY File
Reading PLY: 66826 verts... 133449 faces... done.
Warning: Zero-length normals detected: 0 face normals, 12 vertex normals
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 265
NVM: Number of features: 0
        Loading 100%... done. (Took 571.993s)                                   
Building adjacency graph: 
        Adding edges 100%... done. (Took 0.344s)                                
	200085 total edges.
View selection:
	Building BVH from 133449 faces... done. (Took: 174 ms)
        Calculating face qualities 100%... done. (Took 149.149s)                
        Postprocessing face infos 100%... done. (Took 0.298s)                   
	Maximum quality of a face within an image: 63541.5
	Clamping qualities to 6577.21 within normalization.
	Writing data cost file... done.
	Optimizing:
		Time[s]	Energy
		0	97756
		0	95694
		0	93721
		1	92617
		1	92123
		1	91818
		1	91639
		2	91578
		2	91475
		2	91385
		2	91337
		2	91307
		2	91277
		2	91247
		2	91228
	17096 faces have not been seen
	Took: 153.834s
Generating texture patches:
	Running... Killed
Traceback (most recent call last):
  File "/code/run.py", line 56, in <module>
    app.execute()
  File "/code/stages/odm_app.py", line 92, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 373, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 373, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 373, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 373, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 373, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 373, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 373, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 354, in run
    self.process(self.args, outputs)
  File "/code/stages/mvstex.py", line 97, in process
    '-n {nadirWeight}'.format(**kwargs))
  File "/code/opendm/system.py", line 76, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
 
I processed a set of images collected earlier, and the same installation of opendem produced a map.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I can provide the collection of images if this is not an obvious problem.

skorasaurus, 2019-09-04:
The 137 error is usually the result of running out of memory. 

Regarding the previous set you processed successfully, did that set have fewer images and were its images' dimensions smaller? 

If yes to either, you have a couple different options:

- resize your images to be smaller (before the opendronemap process; there may be a way to do this within odm, i don't  know). 

- decrease the quality of the orthophoto (by increasing[ -orthophoto-resolution ; default is 5](https://docs.opendronemap.org/using.html#command-line-arguments)

- use the[ split-merge approach ](https://community.opendronemap.org/t/child-returned-137/1476)

- obtain more RAM seems to be the most popular answer that i've come across. 

(I'm a newer user to opendronemap so take these with a grain of salt; since this sounds to be more of an issue with your data set, the lead developers encourage you to ask in community.opendronemap.org)

pjmaechling, 2019-09-04:
I tried processing fewer images, and this time it completed without problems, so the out of memory explanation appears correct. Thank you for the specific suggestions, and alternatives. I'll move to the community.opendronemap.org site for these more general questions.

closed 2019-09-04 by pjmaechling

mcorner, 2019-09-04:
If you are using Docker desktop (looks like a MAC) make sure you raise it's memory limit (and probably CPU as well).  The default is often pretty small (like 2GB).