OpenDroneMap/ODM #1024: Incremental image processing

rligocki, 2019-09-11:
I would like to ask if there is any way how to process images on the go. I have a computer, that captures images as a UAV fly survey. My goal is to process images one by one as they are captured so that after a flight, there is no need to post-process anything because orthomosaic or DEM is already generated. Best way to do that would be to create a task and then add images to queue. Everything using python.

Thanks 
Roman

pierotofy, 2019-09-11:
Currently ODM does not support "live" image stitching, although it would be a really cool feature.

We'd welcome contributions to implement this.

Relevant: https://piero.dev/2018/08/toward-real-time-drone-mapping/

annesteenbeek, 2020-07-20:
This is also something I would like to implement. 
Preferably  I would like to be able to generate a task that either monitors a folder or has some hook to take in new images. Maybe also some way to broadcast on when the map has been updated.
I'm just starting to familiarize myself with the codebase, how much work would this be to implement, and does anyone already have some tips on where to start?

If I have time to spare, I would also like to look at supporting this feature into webODM.

Preferably I would have some pipeline that does stitching, DSM, and classification.

pierotofy, 2021-04-05:
This is still a very much welcomed feature, but probably belongs to a separate repository / application for the ODM ecosystem. The methods to do live "stitching" are quite different than the ones used by ODM.

It would be cool if somebody figured out a nice way to package https://github.com/ethz-asl/aerial_mapper which seems to do exactly this.

closed 2021-04-05 by pierotofy

smathermather, 2021-04-05:
Agreed. It is sufficiently different an approach that it would need to be a separate tool.