OpenDroneMap/ODM #1028: [Feature Request] Corner Point Coordinates for input images

ricky1192, 2019-09-17:
It would be great if there was a way to get the coordinates of the 4 corners for the input images at the end of the process, so that we could also trace the input images in a single way.
Thanks,
Riccardo

shreyasbapat, 2019-10-07:
Hi @ricky1192 
Can you point me to some documentation or code file where this can be solved. The issue descriptions is a little vague.

pierotofy, 2021-04-05:
I'm closing this as there has been no interest in implementing this (and no contributions).

For those interested in contributing a PR, one can use the inverse colinearity equations (as implemented in https://github.com/OpenDroneMap/ODM/blob/master/contrib/orthorectify/orthorectify.py#L253) to derive the X/Y geo-coordinates (assuming a flat Z plane).

closed 2021-04-05 by pierotofy

smathermather, 2021-04-05:
I was going to add that many of these use cases may be addressed by the orthorectify module anyway. https://github.com/OpenDroneMap/ODM/pull/1243