OpenDroneMap/ODM #1030: Do we have GPU acceleration now?(2019-09-26)

fclof, 2019-09-26:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions and discussion please open a topic on http://community.opendronemap.org/c/opendronemap. 

Please use the format below to report bugs and faults. It will help improve the resolution process. 
****************************************

### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Ubuntu 18.04

### What is the problem?

Can we utilize NVIDIA GPUs to accelerate the image stitching process? I noticed that we didn't have GPU support back in 2018, but do we have it now, or will we have it in the near future?

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

N/A

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

N/A

pierotofy, 2019-09-26:
Let's move the conversation over to https://community.opendronemap.org @fclof  +1 since this looks like a question.

closed 2019-09-26 by pierotofy

smathermather, 2019-09-26:
Not yet. This is a much requested feature, but to date, no one has come forward with funding or had the time to implement it.

My understanding was that the SIFT patent was expired this year, a patent which applies to much of the free and open source code that is GPU enabled that would be easiest to quickly integrate. That said, [a quick patent search on google ](https://patents.google.com/patent/US6711293) indicates the expiration is more like Spring of 2020.

Piero's right, of course. This is a better conversation for the forum.