OpenDroneMap/ODM #1032: error in dataset spliting

davidcboga, 2019-09-30:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions and discussion please open a topic on http://community.opendronemap.org/c/opendronemap. 

Please use the format below to report bugs and faults. It will help improve the resolution process. 
****************************************

### How did you install OpenDroneMap? (Docker, natively, ...)?

[Docker]

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[windows]

### What is the problem?

[when I run  docker run -it --rm -/docker run -it --rm -v /ortorectificacion/prueba/geo:/code/odm_georeferencing -v /4CCOLOR3A:/code/images -v /ortorectificacion/prueba/orto:/code/odm_orthophoto  odm_image --split 20 execution crash whit this log:
[INFO]    Aligning submodels...
E0930 16:12:42.009611     1 trust_region_minimizer.cc:72] Terminating: Residual and Jacobian evaluation failed.
[INFO]    /code/submodels/submodel_0000/opensfm/reconstruction.aligned.json is now /code/submodels/submodel_0000/opensfm/reconstruction.json
[INFO]    /code/submodels/submodel_0001/opensfm/reconstruction.aligned.json is now /code/submodels/submodel_0001/opensfm/reconstruction.json
[INFO]    /code/submodels/submodel_0002/opensfm/reconstruction.aligned.json is now /code/submodels/submodel_0002/opensfm/reconstruction.json
[INFO]    /code/submodels/submodel_0003/opensfm/reconstruction.aligned.json is now /code/submodels/submodel_0003/opensfm/reconstruction.json
[INFO]    ========================
[INFO]    Processing submodel_0000
[INFO]    ========================
[INFO]    running /code/run.py --project-path /code/submodels --orthophoto-cutline --dem-euclidean-map --skip-3dmodel submodel_0000
/bin/sh: 1: /code/run.py: not found
Traceback (most recent call last):
  File "/code/run.py", line 57, in <module>
    app.execute()
  File "/code/stages/odm_app.py", line 92, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 370, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 351, in run
    self.process(self.args, outputs)
  File "/code/stages/splitmerge.py", line 150, in process
    system.run(" ".join(map(quote, argv)), env_vars=os.environ.copy())
  File "/code/opendm/system.py", line 76, in run ]

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

[Type answer here]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[Type answer here]

pierotofy, 2020-09-14:
Hey @davidcboga :hand: I know this report is a bit old, does this problem still occur with the latest version? We've made a lot of improvements since. :pray:

pierotofy, 2021-10-15:
Closing, assuming fixed.

closed 2021-10-15 by pierotofy