OpenDroneMap/ODM #1035: Add option to generate MBTiles

pierotofy, 2019-10-02:
As part of the `postprocessing.sh` simplification changes that are underway, NodeODM will no longer automatically generate MBTiles. This task should be performed in ODM (via an optional flag), if MBTiles are needed.

This is a matter of running:

```
gdal_translate $orthophoto_path $orthophoto_mbtiles_path -of MBTILES
gdaladdo -r bilinear $orthophoto_mbtiles_path 2 4 8 16 32 64 128 256 512 1024 2048 4096 8192 16384
```

This might affect future POSM builds? (FYI @mojodna )

pierotofy, 2020-09-11:
Closing since nobody seems to have brought forward this problem.

closed 2020-09-11 by pierotofy

quietlychris, 2024-04-07:
Apologies for re-opening dead thread, but this is a feature that I would be interested in seeing made available, particularly if there was an option transferred up to WebODM for configuring pyramids/exporting orthophotos. I've been using a variation on the above suggested GDAL-based commands for a while, especially since high-resolution UAV data doesn't play well with `gdal_translate`'s max zoom level of 22 per https://github.com/OSGeo/gdal/issues/3126, there's a currently several other interim processing steps that are a little clunky that I need to manually tune to end up with a usable MBTiles output.

ifzm, 2024-07-19:
How to adjust, can you give some suggestions? I am also facing high-resolution problems now @quietlychris Thanks )