OpenDroneMap/ODM #1045: ClusterODM “[WARNING] LRE: submodel_0001 failed with”

rumenmitrev, 2019-10-28:
Its not working somehow.
Sub models are only distributed to local node even that ClusterODM has 2 online nodes(1 local /1 remote).
And it fails with :slight_smile: [WARNING] LRE: submodel_0002 failed with: [{“name”:“verbose”,“error”:“Cannot convert 9a3aa963-a5f9-4742-a589-907945610be5 to boolean”}]

console output:
<details>
  <summary>Click to expand</summary>
[INFO]    Initializing OpenDroneMap app - Mon Oct 28 08:24:10  2019
[INFO]    ==============
[INFO]    build_overviews: False
[INFO]    camera_lens: auto
[INFO]    cameras: {}
[INFO]    crop: 3
[INFO]    debug: True
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 2.0
[INFO]    depthmap_resolution: 1000.0
[INFO]    dsm: True
[INFO]    dtm: True
[INFO]    end_with: odm_orthophoto
[INFO]    fast_orthophoto: False
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    ignore_gsd: True
[INFO]    matcher_distance: 0
[INFO]    matcher_neighbors: 8
[INFO]    max_concurrency: 4
[INFO]    merge: all
[INFO]    mesh_octree_depth: 12
[INFO]    mesh_point_weight: 4
[INFO]    mesh_samples: 1.0
[INFO]    mesh_size: 300000
[INFO]    min_num_features: 14000
[INFO]    mve_confidence: 0.6
[INFO]    name: 9a3aa963-a5f9-4742-a589-907945610be5
[INFO]    opensfm_depthmap_method: PATCH_MATCH
[INFO]    opensfm_depthmap_min_consistent_views: 3
[INFO]    opensfm_depthmap_min_patch_sd: 1
[INFO]    orthophoto_compression: DEFLATE
[INFO]    orthophoto_cutline: False
[INFO]    orthophoto_no_tiled: False
[INFO]    orthophoto_resolution: 5
[INFO]    pc_classify: True
[INFO]    pc_csv: False
[INFO]    pc_ept: False
[INFO]    pc_filter: 2.5
[INFO]    pc_las: False
[INFO]    project_path: /www/data
[INFO]    rerun: None
[INFO]    rerun_all: False
[INFO]    rerun_from: None
[INFO]    resize_to: -1
[INFO]    skip_3dmodel: False
[INFO]    sm_cluster: True
[INFO]    smrf_scalar: 1.25
[INFO]    smrf_slope: 0.15
[INFO]    smrf_threshold: 0.5
[INFO]    smrf_window: 18.0
[INFO]    split: 100
[INFO]    split_overlap: 80.0
[INFO]    texturing_data_term: gmi
[INFO]    texturing_keep_unseen_faces: False
[INFO]    texturing_nadir_weight: 16
[INFO]    texturing_outlier_removal_type: gauss_clamping
[INFO]    texturing_skip_global_seam_leveling: False
[INFO]    texturing_skip_hole_filling: False
[INFO]    texturing_skip_local_seam_leveling: False
[INFO]    texturing_skip_visibility_test: False
[... 147 lines cut ...]
[INFO]    LRE: About to process submodel_0004 remotely
[WARNING] LRE: submodel_0004 failed with: [{"name":"verbose","error":"Cannot convert 9a3aa963-a5f9-4742-a589-907945610be5 to boolean"}]
[INFO]    LRE: Re-queueing submodel_0004 (retries: 4)
[INFO]    LRE: About to process submodel_0001 remotely
[INFO]    LRE: Waiting 40 seconds before processing submodel_0001
2019-10-28 08:25:23,617 DEBUG: Found 18620 points in 7.92439603806s
2019-10-28 08:25:23,620 DEBUG: No segmentation for 20181221_Miaoli_Shitan_ms2_074.JPG, no features masked.
2019-10-28 08:25:23,848 INFO: Extracting ROOT_HAHOG features for image 20181221_Miaoli_Shitan_ms1_111.JPG
2019-10-28 08:25:25,596 DEBUG: Found 18747 points in 8.14708900452s
2019-10-28 08:25:25,598 DEBUG: No segmentation for 20181221_Miaoli_Shitan_ms3_71.JPG, no features masked.
2019-10-28 08:25:25,740 INFO: Extracting ROOT_HAHOG features for image 20181221_Miaoli_Shitan_ms3_84.JPG
2019-10-28 08:25:30,005 DEBUG: Found 18434 points in 8.06343984604s
2019-10-28 08:25:30,006 DEBUG: No segmentation for 20181221_Miaoli_Shitan_ms2_052.JPG, no features masked.
2019-10-28 08:25:30,268 INFO: Extracting ROOT_HAHOG features for image 20181221_Miaoli_Shitan_ms2_113.JPG
2019-10-28 08:25:31,894 DEBUG: Found 17620 points in 7.92464804649s
2019-10-28 08:25:31,896 DEBUG: No segmentation for 20181221_Miaoli_Shitan_ms1_111.JPG, no features masked.
2019-10-28 08:25:32,140 INFO: Extracting ROOT_HAHOG features for image 20181221_Miaoli_Shitan_ms3_65.JPG
</details>

[image]

[image]

engine version 0.9.1
Api version 1.5.3

pierotofy, 2019-10-28:
Thanks for the report @rumenmitrev :+1:

If you remove the `--verbose` flag, does it process correctly?

bertramt, 2019-10-28:
I'm running into that issue too

> : [{"name":"dsm","error":"Cannot convert fd124845-8ec3-43b4-93d0-168a043fd990 to boolean"}]

with the options 

> sm-cluster: http://192.168.0.165:3001, rerun-from: dataset, split: 500, dsm: true

pierotofy, 2019-10-28:
Could you detail your setup? 

 - Is the task being sent from WebODM? (I think yes, but just want to make sure)
 - Are you sending the task to a ClusterODM or NodeODM node?

rumenmitrev, 2019-10-28:
Yes. The task is initiated through WebODM.
At this machine resided also ClusterODM and NodeODM

On the other machine is only NodeODM

pierotofy, 2019-10-28:
When you send a task for processing, do you send it to ClusterODM or NodeODM?

rumenmitrev, 2019-10-28:
sm-cluster: http://127.0.0.1:3001
is ClusterODM

bertramt, 2019-10-28:
I initially tried to add the cluster as a processing node and for some reason it didn't work come online so I assumed that setting the sm-cluster option overrides the node.  I back and restarted everything and now was able to add my cluster as a node.  Trying to reprocess now.

rumenmitrev, 2019-10-29:
if I initiate a task via ./odm command (CloudODM) no errors .
Models are distributed correctly to remote host and task is running fine.
But no joy without GUI :)

wang1212, 2019-11-08:
I ran into the same problem, I tried it on **ubuntu 18.04 server**. And I create a task through **WebODM** and add **ClusterODM** to Process Nodes. 

I processed 136 images and the options for the error were:

```
sm-cluster: http://192.168.1.120:3001, rerun-from: dataset, debug: true, split-overlap: 5, split: 70, fast-orthophoto: true, max-concurrency: 1
```

I tried setting `verbose` to `true`, but it caused the error to occur, then I set it to `false` and tried to set `split-overlap` to `3`, `10`, `15`, still not solved.

In the end, I succeeded, but I don't know the reason at the moment. The options are as follows:

```
sm-cluster: http://192.168.1.120:3001, rerun-from: dataset, debug: true, split-overlap: 50, split: 70, fast-orthophoto: true, max-concurrency: 1
```

I just set `split-overlap` to `50` and the error disappears.

If anyone else meets the same problem, can you give a solution and a specific reason? Thank you!

pierotofy, 2020-09-11:
These issues should have been solved with https://github.com/OpenDroneMap/ODM/commit/59edf1fd0f63551470e118c87817f0d768b2dbc0 a few months ago. Please re-open if the issue persists? :pray:

closed 2020-09-11 by pierotofy