OpenDroneMap/ODM #1051: Opendronemap 0.9 processing speed goes down as compare to 0.5

katarias, 2019-11-06:
Hi All,

I have recently installed ODM 0.9 in my ubuntu server. 
CPU - 64 Core
RAM - 256 G
Previously i was using ODM 0.5. 
ODM 0.9 working fine with upto 200 images it takes approx same time as ODM 0.9 and GCP result is better.  But for large dataset like 400/500 images it takes too much time.

This is what is measure while dataset was in process - 
- 450 images ODM 0.5 took approx 5 hrs where ODM 0.9 took 7 hrs.
- Most of processing time was happening on opensfm match_features function
- Server has 64 Core CPU and server load crosses 1000 when it runs match features.
- it running N(no. of cpu core) number of python loky processes in backend which exhausted all CPU resources.
This service running in backend as part of match features - 
         /usr/bin/python -m joblib.externals.loky.backend.popen_loky_posix --process-name LokyProcess-64 --pipe 65 --semaphore 6
- Tried to change --max-processes for concurrency but still has approx same result.

If anyone faced this issue with ODM 0.9 or any idea on that please let me know.

Thanks

pierotofy, 2020-09-11:
Hey @katarias :hand: please try the latest version. You might find that we have improved speed in many areas. I'll close this for now. :pray:

closed 2020-09-11 by pierotofy

katarias, 2020-09-12:
Thanks @pierotofy . Using latest ODM 1.0.2 which has really improved speed.