OpenDroneMap/ODM #1054: Better GCP file validation

pierotofy, 2019-12-10:
Sometimes users are confused about how to create a GCP file, which can lead to a GCP file with not sufficient entries, or with entries that don't reference enough images (< 2).

ODM should be more forgiving and give some hints as how to fix the problem.

jeffin07, 2020-10-24:
@pierotofy is this still open? If so can you give me some guidance on this issue as I'm new to this project? I thought a better way to understand the project will be to try to fix beginner-friendly issues

pierotofy, 2020-10-24:
Yep this one is still open.

Get familiar with using GCPs (https://docs.opendronemap.org/gcp.html#gcp-file-format), then what we're trying to improve is reporting better error messages to users when they provide an invalid GCP file.

pierotofy, 2022-03-08:
Closing, current logic is sufficient.

closed 2022-03-08 by pierotofy