OpenDroneMap/ODM #1066: Upgrade to Python 3

pierotofy, 2020-01-24:
Python 2 is at the end of support. We should update the codebase to Python 3 (probably at the same time as upgrading to the latest Ubuntu LTS).

skywalkerisnull, 2020-03-01:
Has anyone tried to run 2to3.py over the project and seeing what happens? I ran it from a windows machine but it gave out some odd formatting (double line breaks everywhere).

pierotofy, 2020-03-01:
I haven't, but migration to Python 3 is something we really need to do (along with #834 ). I know the guys at SolSpec started a port long time ago, but it was never integrated into master so some modules might be missing (but could be a good reference). https://github.com/SolSpecSolutions/ODM

ABitMoreDepth, 2020-04-22:
I don't know this codebase, but would potentially be willing to try and work on this, if you're open to PRs?

pierotofy, 2020-04-22:
Absolutely! We'd welcome any effort to move this to Python 3.

ABitMoreDepth, 2020-04-22:
As I'm fresh to the ODM project (but not to python), would you have any suggestions on where I should start in terms of playing with code, which codebase to start with etc?

pierotofy, 2020-04-22:
I'd start by setting up a dev environment (https://github.com/OpenDroneMap/ODM#developers), installing a Python 3 virtualenv, run the `pip install -r requirements.txt`, then run the pipeline via `./run.sh [...]` and start from the first module/stack trace that fails.

Note the `./start-dev-env.sh` can work well in OSX too, but needs a few tweaks. If you're on Windows, see how the script launches the docker container (the important part).

jeffin07, 2020-10-24:
@pierotofy I would like to work on this issue. Any heads-up?

pierotofy, 2020-10-24:
Hey @jeffin07 :hand: thanks for the offer! I actually forgot to close this issue as we migrated to Python3 with #1156 already!

There are other issues that perhaps you can help, take a look at this list: https://github.com/OpenDroneMap/ODM/issues?q=is%3Aopen+is%3Aissue+label%3A%22help+wanted%22

closed 2020-10-24 by pierotofy