OpenDroneMap/ODM #1067: Multispec - Process exited with code 1 Error

JCAAUSFRA, 2020-02-04:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Edge

### What is the problem?

tried to process multi spec from Altum. got a "Process exited with code 1" error

JCAAUSFRA, 2020-02-10:
Edge 44 on Windows 10

pierotofy, 2020-09-11:
@JCAAUSFRA can you share your dataset and/or a copy of the output task? I know this issue is pretty old, but it's going to be though to solve without a dataset to test on or more info.

JCAAUSFRA, 2020-09-12:
https://infarmau-my.sharepoint.com/:f:/g/personal/jerome_leray_infarm_io/EoMPtqKVzeVOsN8Wn1Q9XoYBGV8YzzjvalWiCJ_lAktGjQ?e=e6P9sR

pierotofy, 2020-09-15:
I'll try to look into this one of these days.

pierotofy, 2020-09-16:
The problem is due to the thermal band, which has a very small image size.

ODM needs better support for such images and will currently fail.

A workaround if you don't need the thermal band is to remove all the thermal band images.

pierotofy, 2021-04-05:
Altum cameras are now supported! :clinking_glasses:

closed 2021-04-05 by pierotofy