OpenDroneMap/ODM #1070: Orthophoto of multispectral

KeoChi, 2020-02-12:
### How did you install OpenDroneMap? (Docker, natively, ...)?

natively

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

ubuntu16.04 

### What is the problem?
 I have run the newest code with [MicaSense dataset](https://community.opendronemap.org/t/micasense-red-gren-blue-near-ir-red-edge/3176). I found that I use "Image Viewer" to open the "odm_orthophoto.original.tif" and the garbled characters appeared on the right side of the image:
[image]
  In addition, I use opencv to imread and imshow\imwrite the "odm_orthophoto.original.tif" that the result is same.
  Do the odm_orthophoto module have bug?

pierotofy, 2020-02-12:
Possibly, this might need to be investigated.

KeoChi, 2020-02-14:
It look normaly in "Global Mapper". I have no QGIS.

pierotofy, 2020-02-16:
https://www.qgis.org/en/site/

If it looks fine in both Global Mapper and QGIS, then it might be a problem with image viewer. The resulting file has 5 bands + alpha, which might be incorrectly handled by the image viewer.

KeoChi, 2020-02-18:
@pierotofy Hello. I found that the alpha can influence the resulting file. But I don't know the reason.

pierotofy, 2020-09-11:
Closing since this looks like a problem with the image reader, not ODM. :pray:

closed 2020-09-11 by pierotofy