OpenDroneMap/ODM #1071: [Feature] Scale constraints support

pierotofy, 2020-02-13:
It has been suggested that the software could add scale constraints as described here https://community.opendronemap.org/t/scale-constraint/3622/4

This might need a different type of annotation file (similar to GCPs).

Saijin-Naib, 2020-02-13:
What would be needed to describe each scale constraint?
XYZ and maybe a "constraint ID", so we know which two manual tie points are being used as a scale constraint together?

pierotofy, 2021-04-26:
Being investigated at https://github.com/mapillary/OpenSfM/issues/737

pierotofy, 2021-04-26:
Yes.

pierotofy, 2022-12-28:
This is tricky; points from a 2D image could be triangulated to a closest 3D points in the **dense** point cloud, then the 3D points could be back-projected to the 2D cameras and manual features could be added to the feature list. Then (a modified) OpenSfM would have to be re-run using these novel features/3D point constraints.

And what if for any reason there are no points nearby?

An easier approach could be to apply an affine transform based on the closest 3D points alone, expect that I wouldn't trust it's accuracy due to camera angle (it would only be accurate when shots are perfectly nadir to the measurement surface).

There's a reason other software makes this process interactive and based on 3D measurements.

I'm going to close this as being out of scope. People can interactively scale a model using CloudCompare as a post-process operation.

closed 2022-12-28 by pierotofy